greatscottgadgets/hackrf #155: IQ Data form usb is signed?

wzyy2, 2015-01-05:
If I google "hackrf iq",I will be told that "The HackRF logs are IQ data pairs, stored as unsigned 8 bit integers "..............
even comment in gr-osmosdr say is unsigned.....
```
/* HackRF device produces 8 bit unsigned IQ data */
...........
    _lut.push_back( gr_complex( (float(char(i >> 8))) * (1.0f/128.0f),
                                (float(char(i & 0xff))) * (1.0f/128.0f) ) );
...........
```
Maybe It should be mentioned.

mossmann, 2015-04-08:
The data are signed but were once (more than a year ago) unsigned.

a232319779, 2017-01-22:
HI,mossmann,
    which firmware version had been changed?
    How about the version : 2014.08.01

a232319779, 2017-01-22:
I should ask clearly:
it's signed at the firmware version 2014.08.01?

dominicgs, 2017-01-22:
Yes, the change to signed data was introduced in release 2014.08.1.

closed 2017-01-25 by mossmann