greatscottgadgets/hackrf #162: Ubuntu 14.04 Cmake fails with LIBUSB_INCLUDE_DIR not set

misham, 2015-03-06:
I have libusb and libusb-dev installed.  I tried this with `pybombs` and directly from the repo.

If `LIBUSB_INCLUDE_DIR` is not defined, Cmake fails with the following:

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBUSB_INCLUDE_DIR (ADVANCED)
```

I fixed the build with setting `LIBUSB_INCLUDE_DIR=/usr/include/libusb-1.0` when running `cmake`

dominicgs, 2015-04-07:
This is caused by not having pkg-config installed, which prevents cmake from finding libusb.

I'll take a look at updating the pyBombs recipe.

closed 2017-11-17 by dominicgs

Serjei334, 2024-10-11:
**_sudo apt -y install pkg-config_**
It helped me