greatscottgadgets/hackrf #163: Detach the kernel driver automatically

ao2, 2015-03-25:
Hi,

I saw some complaints about the kernel driver interfering with libhackrf:
https://twitter.com/michaelossmann/status/579303581829988352/photo/1

There is a fairly easy solution to that, also proposed by the kernel driver developer: let libusb detach the kernel driver.

Here is a patch which can be applied with ```git am```:
http://ao2.it/tmp/0001-Automatically-detach-reattach-the-kernel-driver-if-l.patch

I only compiled-tested the patch, I don't have a hackrf myself.

if you want you can amend it to use ```true``` instead of ```1``` but I used ```1``` because that's what libusb expects.

Ciao,
   Antonio

dominicgs, 2015-04-07:
Hi Antonio,

This patch looks great, I'm in the process of spinning up a VM to test it, but I don't anticipate any problem with it.

Thanks,
  Dominic

dominicgs, 2015-04-07:
I have performed some testing on this patch.  Unfortunately, for the system that I am testing (Kali with 3.18), it does not work.

For a reason which is not yet clear, the ```libusb_set_auto_detach_kernel_driver``` function returns ```LIBUSB_SUCCESS``` but we are still unable to claim the interface or set the configuration.

If I replace the call to ```libusb_set_auto_detach_kernel_driver``` with a pair of calls to ```libusb_kernel_driver_active``` and ```libusb_detach_kernel_driver```, it works as expected.  I am not yet certain why this works when the automatic method fails.

Additionally, the ```LIBUSB_VERSION``` define is a product of the libusbx fork, so for my version it is named ```LIBUSBX_VERSION```, I assume that this will not change in future versions.

ao2, 2015-04-08:
libusb and libusbx have been merged back together (http://libusb.info), the ```LIBUSBX_API_VERSION``` name is now used just for compatibility and possibly it will be deprecated in the future:
https://github.com/libusb/libusb/blob/master/libusb/libusb.h#L146

Obviously if you have an older libusb which doesn't have ```LIBUSB_API_VERSION``` the code from my patch will not be compiled in, but I take you changed it locally to use ```LIBUSBX_API_VERSION```.

By looking at the libsub code ```libusb_set_auto_detach_kernel_driver``` is supposed to always return ```LIBUSB_SUCCESS``` on platforms which support kernel driver detaching, that function just sets a flag, the detaching is performed at claim_interface-time.
Calling  ```libusb_claim_interface()``` with the auto detach flag set should be equivalent to the explicit detaching done before ```libusb_claim_interface()```, so I don't know either what's going on.

You can try adding something like this:
```diff
diff --git a/host/libhackrf/src/hackrf.c b/host/libhackrf/src/hackrf.c
index 1d1d828..72311f8 100644
--- a/host/libhackrf/src/hackrf.c
+++ b/host/libhackrf/src/hackrf.c
@@ -271,6 +271,8 @@ int ADDCALL hackrf_open(hackrf_device** device)
                return HACKRF_ERROR_INVALID_PARAM;
        }

+       libusb_set_debug(g_libusb_context, 4); /* LIBUSB_LOG_LEVEL_DEBUG */
+
        // TODO: Do proper scanning of available devices, searching for
        // unit serial number (if specified?).
        usb_device = libusb_open_device_with_vid_pid(g_libusb_context, hackrf_usb_vid, hackrf_one_usb_pid);
```

Just curious, what version of libusb/libusbx are you using?

Anyways, I can re-send a version with the two explicit calls, skipping the version check, if that works for you.

Ciao,
   Antonio

dominicgs, 2015-04-08:
@ao2 You're right, both are defined from 1.0.18 onwards, unfortunately debian/ubuntu/kali all ship 1.0.17.  Most users who see this problem are running kali.  I think checking for both is probably a good idea, given that we are likely going to have to span the time when ```LIBUSBX_VERSION``` is removed and some people are still running code without ```LIBUSB_VERSION```.

I'm also concerned about the order of calls to libusb.  First we open the device, then we set the configuration and then we claim the interface.  Can we set the configuration if the kernel is attached to an interface?  Having added in some debug logging, it appears as though ```libusb_set_configuration``` is returning ```LIBUSB_ERROR_BUSY```.

ao2, 2015-04-08:
So was it just that the code was not compiled in? Does the auto detach feature itself actually work?

If so, I can prepare a patch which checks both symbolic constants.

About when to call ```libusb_set_configuration``` I though about that too, I am going to ask libusb developers about it. Problem is that the kernel driver detaching is currently performed when claiming the interface, if it turns out it should be performed before setting the configuration the auto detach mechanism as implemented in current/older libusb is not going to work as it should, and we'd be better off adding the explicit detach before setting the configuration.

Let's see what libusb devs say first.

dominicgs, 2015-04-08:
The code wasn't being compiled in, but once I had modified the ifdef then it was being compiled in.  However, it still wasn't working.  I'm fairly sure that this is because we try to set the configuration before claiming the interface.  As you say, the kernel driver is detached when we claim the interface.

I don't think that you can claim an interface before setting a configuration can you?  Wouldn't that mean that you were claiming an interface which may not exist under the currently active configuration?

I'm looking forward to hearing what the libusb devs suggest.

dominicgs, 2015-04-08:
From the libusb mailing list, the order of events seems to be as follows.

For all interfaces on the active configuration:
```libusb_detach_kernel_driver(dev, interface_number);```

Then:
```
libusb_set_configuration(dev, configuration);
libusb_set_auto_detach_kernel_driver(dev, 1);
libusb_claim_interface(dev, interface_number);
```

ao2, 2015-04-10:
Yes, this would be the most general solution.

However by knowing the hardware and the kernel driver maybe we can simplify things and just get away with it by calling ```libusb_set_configuration()``` conditionally, as shown in http://libusb.sourceforge.net/api-1.0/caveats.html#configsel

The rationale is that, if the hardware has only one useful configuration:
* if there is a kernel driver loaded it has set the right configuration already
* if there is no kernel driver ```libusb_set_configuration()``` will succeed

I'll come up with new tentative patches as time permits.

ao2, 2015-04-20:
HI @dominicgs I only now noticed that you worked on a fix for the issue yourself, but since I happened to have done something myself as well I am posting it anyway just for you to take a quick look:
http://ao2.it/tmp/libhackrf-detach-kernel-driver.patch

I will make more specific comments under the commit itself.

dominicgs, 2015-06-29:
Fixed in pull request #185

closed 2015-06-29 by dominicgs