greatscottgadgets/hackrf #166: Why the processing is limited to 6 GHz?

mouradghafiri, 2015-04-07:
Hello,

First I would like to express my deepest gratitude to the team that is working behind this project.
I have just a special question, I'm wondering why such great solution doesn't offer a flexibility to process in higher bands, for example up to 8 or 10 GHz? if it is so, will it take time to adapt the HackRF's source code to support higher bands.

Best Regards;

dominicgs, 2015-04-08:
@mouradghafiri The 6GHz limit is the highest frequency that the HackRF One hardware can tune to.  For higher frequencies you would need to attach an down-converter to the HackRF, in the same way that the Ham It Up exists for accessing lower frequencies.

mossmann, 2015-04-08:
a little more information on why the 6 GHz limit exists: The hardware architecture requires several components, including a mixer and some RF switches, that can operate over the entire frequency range from 1 MHz to 6 GHz.  We were unable to find affordable components that support such a wide range of operating frequencies beyond 6 GHz.  Additionally, losses from the PCB construction itself become quite high above 6 GHz without switching to a more expensive PCB substrate.

closed 2015-04-08 by mossmann