greatscottgadgets/hackrf #168: Activating "Phantom" Antenna power?

regulatre, 2015-04-09:
Looking for a way to activate phantom antenna power on my HackRF while using it in GQRX, Gnuradio, and other Osmo based applications. Is there a way to activate this feature through hardware means - perhaps a jumper? I have a LNA4All modified for phantom power but I'm struggling to find a way to use it with applications using the Osmo driver. 

Thanks!

dominicgs, 2015-05-11:
@regulatre Thank you for creating this issue.  We support antenna port power from the HackRF command line tools and libhackrf, so this will require an update to [gr-osmosdr](http://git.osmocom.org/gr-osmosdr)

Something that handles power in the same way as "buffers" is already handled by the sink block: http://git.osmocom.org/gr-osmosdr/tree/lib/hackrf/hackrf_sink_c.cc#n172

ckuethe, 2015-05-11:
Airspy also has antenna power, controlled by "bias=1" in the device
arguments. Seems like a place to start

regulatre, 2015-05-13:
Thanks for the hints guys. I'll see what I can come up with.

regulatre, 2015-06-04:
Added! gr-osmosdr Commit ID 594391982845b4ac0e4e48bbb91b4b85552f2af6 May 27th, 2015

closed 2015-06-04 by regulatre

zodoczi, 2017-02-07:
Anyone know how to enable antenna power in Windows, SDR#?