greatscottgadgets/hackrf #169: where i can download the source code of SDR#

commthu, 2015-04-13:
Only nightly builds currently support HackRF One - http://sdrsharp.com/downloads/sdr-nightly.zip

I want to make some modification of SDR# to use hackrf, where i can download the source code. 
Thanks

dominicgs, 2015-04-13:
SDR# is not part of the HackRF project.

This question should be directed to the author(s) of that project, although I do not believe that it is open source software.

closed 2015-04-13 by dominicgs

commthu, 2015-04-13:
In fact,  i do not need SDR# source code.
I want to make some modification of hackrf.dll  where SDR# used.  Can i get the source code of hackrf.dll in SDR#.

dominicgs, 2015-04-13:
The hackrf.dll file is a Windows build of libhackRF, which can be found here: https://github.com/mossmann/hackrf/tree/master/host/

Using cmake and your compiler of choice on Windows should allow you to build libHackRF as a DLL.

commthu, 2015-04-13:
I complied that last week, it only generate 13kb size dll, however, SDR#
use 78kb size dll,  I replaced it, then SDR# failed to open.

miek, 2015-04-13:
Here's the source for the SDR# HackRF plugin: https://github.com/zefie/sdrsharp_hackrf

commthu, 2015-04-13:
Thanks miek and dominicgs.