greatscottgadgets/hackrf #172: Does Hackrf support GPS time-stamping the RF data?

commthu, 2015-04-20:
I want to use GPS time stamping the RF data, does hackrf support this function.
I know some SDR device has this function. This is very useful in some applications.

KC5CQW, 2015-08-06:
This will probably require additional hardware like the PortaPack H1 and a GPS to sync with its RTC. 
It may be possible to stream NEMA outputs into a GPIO but otherwise, this would seem to be a host software feature.

ckuethe, 2015-08-06:
I wrote a gnuradio module to ingest NMEA data into your flowgraph. You could use that to write stream tagged samples to a data file... https://github.com/ckuethe/gr-nmea

KC5CQW, 2015-08-06:
Boom Shaka Lahka. 
And there it is...

ckuethe, 2015-08-06:
Note that this module can't push any timing information into the receiver
so you'll probably get some pretty jittery timestamps depending on system
load.

mossmann, 2017-01-26:
Thanks, ckuethe! I'm closing this issue as the remaining concern (timestamps implemented in firmware instead of on the host) already has an issue: #85

closed 2017-01-26 by mossmann