greatscottgadgets/hackrf #174: HackRF and L-Band Spectrum

mouradghafiri, 2015-05-04:
Hello,
Please is it possible to use HackRF and GNU Radio to plot the spectrum of the whole L-Band (1 to 2 GHz) assuming we design an antenna that could capture this band.

Thank you in advance

ckuethe, 2015-05-06:
Yes. HackRF is specified to work from 10MHz to 6GHz with no gaps.

mouradghafiri, 2015-05-06:
I'm trying to order the hackrf but I need to know some details before
choosing it as a condidate.
What about the legal framework of it, I mean anyone could buy it regardless
of the regulatory boundries.

One last question about the 20 MHz of bandwith limits!, I don't understand
how it can capture the whole L-band and has 20MHz band limit?

Would you please elaborate a bit.

Regards;

ckuethe, 2015-05-06:
You can tune anywhere between 10 and 6GHz (even more, actually) and sample up to +/- 10MHz from that frequency. That's enough to grab the entire commercial FM radio band, not enough to tune GPS L1/2/5 (for example). You could write a program to scan the band in 20MHz chunks, but HackRF won't be able to grab 1GHz of spectrum all at once.

mouradghafiri, 2015-05-06:
Thank you so much, it is clear enough now.

closed 2015-05-08 by dominicgs