greatscottgadgets/hackrf #175: Could I use two or more hackrf ones with one host at the same time?

franklinzhang, 2015-05-08:
I've tried in my computer(os: ubuntu 14.04) but failed.

dominicgs, 2015-05-08:
This feature has been implemented but is, as yet, untested and therefore has not been released.  The code for it is contained in theis pull request: https://github.com/mossmann/hackrf/pull/158

If you would like to use it, I would recommend that you try that version of the host software.  Feedback on how it works for you would be very welcome.

kpansky, 2015-05-08:
I have been running with #158 applied for some time now, and it works very well.  I have not gone through the effort of building and testing the updated GNU radio osmosdr driver using the updated libhackrf, but the updated command line tools/libhackrf/LPC firmware seem to work well at least.

It would be great to get #158 merged and released along with the hessu/gr-osmosdr updates.

dominicgs, 2015-05-11:
Closing this issue as it is addressed by pull request #158 and is a duplicate of issue #87.

closed 2015-05-11 by dominicgs