greatscottgadgets/hackrf #180: Firmware should stop TX/RX when communication is lost

kpreid, 2015-06-09:
Feature request:

If the host process which is controlling the HackRF is abruptly terminated, the HackRF remains in TX or RX mode. It would be nice if the firmware would instead recognize this condition and shut off the transmitter/receiver.

I'm not familiar enough with USB to know whether this is a detectable event (analogous to a lost TCP connection) or would require timeout logic (e.g. not receiving any TX data packets or other keepalive).

KC5CQW, 2015-08-06:
User adjustable "watchdog" timer is especially for TX is needed for compliance with various communications authorities. Not to mention, equipment/DUT safety.

rroff, 2016-07-13:
Hey,

Have you seen that latest movie? I don't know what all the fuss is about. Before you watch it read a review here <http://tusindutro.peterbillingsley.org/aecpdyn>

See you soon, rroff@roff.us

dominicgs, 2017-11-17:
For TX I plan to keep track of the freshness of the buffers to be transmitted. There's no sense in transmitting buffers that have already been transmitted (not in our current usage model), so I'll reset to idle if that happens.

I'm fairly sure it could work that way for RX, but I don't know if total failure is the answer for a slower than expected USB connection.  I think it's a fair thing to do, but I worry that it will break some things in subtle ways.

closed 2022-03-02 by mossmann