greatscottgadgets/hackrf #181: Random Reboots with HackRF Jawbreaker

alexschneider, 2015-06-17:
Hi All,

I'm trying to use my Jawbreaker with a Thinkpad W530 under Arch Linux. I've installed the hackrf tools and libhackrf from git, and flashed the latest release of the firmware onto my device. About half of the times I use hackrf commands or GNU Radio, it fails with the message `hackrf_open() failed: HACKRF_ERROR_LIBUSB (-1000)`. Further investigation reveals that the reason for this error is because of [this](https://github.com/mossmann/hackrf/blob/master/host/libhackrf/src/hackrf.c#L523) control path. If the command fails, the usb light on the device flashes and I see a usb disconnect/connect in my dmesg. 

I can reliably get it to work by running the following command:

```
$ hackrf_info; sleep 1; hackrf_info
```

(note: the first one will fail, but the second one always succeeds). 

I've tried about a dozen different usb cables (including one with a ferrite core), and all the ports on my computer, with no difference in result.

Manawyrm, 2015-06-25:
I have that issue with my ThinkPad T410 too, it seems to be a powering issue. When powered over a Y-cable from a power bank, the HackRF works smoothly.

dominicgs, 2015-06-25:
@Manawyrm Thanks for the input, it's useful to know that others see the same issue.  Do you still see the issue when the laptop is powered by the AC adapter?  I'm wondering if the ThinkPads have more aggressive power saving settings than other systems?

@alexschneider Are you running the latest firmware?  I know there was a problem with an early Jawbreaker firmware not supporting suspend correctly.  I'm surprised that the command succeeds the second time and that it fails when listing all devices.  Does ```lsusb``` exhibit similar issues?  Or is it only with HackRF tools?

Manawyrm, 2015-06-25:
@dominicgs That's an interesting question. My tests were made on a field day with the hackRF and lapton running on DC/battery. I will try to power it with the AC adapter this evening and report the results..

alexschneider, 2015-06-25:
@dominicgs I was using the latest firmware available from the releases page. Only the HackRF tools exhibit this issue, and the laptop was being powered by an AC adapter.

Manawyrm, 2015-06-25:
The problem also occures with GRCs osmocom sink for me...

Manawyrm, 2015-06-27:
Finally did some testing: 
The problem occures with both DC and AC power.
Using a powered USB hub everything works fine.

dominicgs, 2015-06-28:
@Manawyrm Are you using the latest firmware?  There's a binary in the 2014.08 release.

Could you paste the output from hackrf_info?

Manawyrm, 2015-06-28:
```
Found HackRF board.
Board ID Number: 2 (HackRF One)
Firmware Version: 2014.08.1
Part ID Number: 0xa000cb3c 0x005f4349
Serial Number: 0x00000000 0x00000000 0x14d463dc 0x2f79aee1
```

dominicgs, 2015-06-28:
@Manawyrm This sounds very similar to issue #97.

The workaround is here: http://nine.pairlist.net/pipermail/hackrf-dev/2013-August/000169.html

Manawyrm, 2015-06-28:
@dominicgs Unfortunatly that doesn't solve the problem. I do think that this is related to a power issue. (at least on my T410)

dominicgs, 2015-07-08:
@alexschneider Did you have chance to test whether or not this workaround fixes your issue?

alexschneider, 2015-07-14:
@dominicgs I'm running TLP which has a similar mechanism. That workaround didn't fix my issue, but completely stopping TLP did. I'll just have to remember to do so whenever I want to use my hackrf.

dominicgs, 2015-07-18:
Another user has reported this issue and was able to work around it by disabling auto-suspend for HackRF in ```/etc/default/tlp```.

GrayShade, 2015-09-14:
I encountered this issue because of an `udev` rule enabling USB autosuspend for every device. I added the rule myself, but running `powertop --autotune` (or manually toggling that tunable) has the same effect.

closed 2019-11-25 by miek