greatscottgadgets/hackrf #182: Any example to run M0 core? Just change M0_sleep.c?

JiaoXianjun, 2015-06-24:
Does Current firmware image creation script include M0 image also?
If not, could anyone give me some hints on how to run M0? (at least can control LED via GPIO by M0).

Thanks a lot!

dominicgs, 2015-06-24:
The best example of how the M0 is used is the PortaPack firmware.  Of course, it will not do anything useful without a PortaPack, but it is a good starting place for your own code.

The instructions for building the PortaPack firmware are here: https://github.com/sharebrained/portapack/wiki/Building-and-Updating-Firmware

The firmware is here: https://github.com/sharebrained/portapack/tree/master/portapack_hackrf/firmware/portapack_hackrf

I spoke to @sharebrained about the code and he offered these pointers:
```
The gist is that I start the M4 with a very lightweight bootloader. It starts the M0 from the SPI flash and then sleeps. (The M0 is doing all the UI stuff.)
Then the M0 loads code into RAM and points the M4 at it and resets it.
(The M4 is acting as a baseband DSP, doing very little other than signal processing.)
This way, I can have different modes/applications with their own baseband DSP implementation and code, and can switch it out at runtime from the M0.
```

Hoernchen, 2015-06-24:
There is also my m0 branch, i gave using the m0 a try back in 2013 : https://github.com/Hoernchen/hackrf/tree/m0

JiaoXianjun, 2015-06-25:
Hi Hoernchen,

Seems your work is M0 standalone? Not run with M4 at the same time?

JiaoXianjun, 2015-06-25:
Portapack' method seems so complicated.

Hoernchen, 2015-06-27:
No, it is not standalone, the m4 starts the m0 which handles the sgpio interrupt for reading & writing, the m4 still does everything else.

closed 2015-07-14 by dominicgs