greatscottgadgets/hackrf #190: PyBOMBS install fails without root

miek, 2015-07-08:
The standard instructions for PyBOMBS recommend to install locally in $HOME, but the HackRF install step will always try to install udev rules to /etc so the install fails.

    -- Installing: /etc/udev/rules.d/53-hackrf.rules
    CMake Error at misc/udev/cmake_install.cmake:44 (FILE):
      file INSTALL cannot copy file
      "/home/mike/pybombs/src/hackrf/host/build/misc/udev/53-hackrf.rules" to
      "/etc/udev/rules.d/53-hackrf.rules".
    Call Stack (most recent call first):
      misc/cmake_install.cmake:37 (INCLUDE)
      cmake_install.cmake:39 (INCLUDE)

Also, this is particularly annoying because it fails silently when you don't use the verbose flag with PyBOMBS:

    mike@r5l ~/pybombs $ ./pybombs install hackrf
    Settled on prefix: /home/mike/target
    Initializing environmental variables...
    Loading recipes ...
    /home/mike/target
    Installing packages:
    * hackrf
    Installing from source: hackrf
    Cloning into 'hackrf'...
    remote: Counting objects: 9530, done.
    remote: Compressing objects: 100% (72/72), done.
    remote: Total 9530 (delta 43), reused 0 (delta 0), pack-reused 9458
    Receiving objects: 100% (9530/9530), 24.48 MiB | 1.93 MiB/s, done.
    Resolving deltas: 100% (6658/6658), done.
    Checking connectivity... done.
     
        CC=gcc CXX=g++ cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_INSTALL_PREFIX=/home/mike/target 
    
    Configuring: (100%) [==========================================================]
    Building:    (100%) [==========================================================]
    
        make install
    
    Installing: (100%) [===========================================================]
    mike@r5l ~/pybombs $

dominicgs, 2015-07-09:
I'm not sure what the correct behaviour is here.

Should this be an error that causes the build to fail?
Should we install the udev rule somewhere else if an install prefix is specified?
Should we disable the udev rule by default when an install prefix is specified?

We don't have a way to test the write permissions on /etc/udev/rules.d until we run ```make install``` because that's the only time that we should be root.  Unless we add an extra "are we root?" test to cmake because pybombs runs everything as root or everything as the user?

I don't believe any of the other pybombs recipes for hardware devices deal with this correctly either.  Is this perhaps a pybombs issue?

ckuethe, 2015-07-09:
I don't think it should be a hard error to fail to install udev rules. It's quite possible that a user will never have a hackrf, but libosmosdr or something like that links against libhackrf just in case.

there are a couple of knobs in cmake to control udev rule installation, but it's not super easy for a pybombs user to pass those in to the hackrf build... i'm not even sure if that's possible without editing the recipes.

dominicgs, 2015-07-09:
Good point, a hard error is not acceptable in that scenario.

I would be happy to submit a modified recipe to pybombs that changes the behaviour, although this isn't just about pybombs.  The same questions arise if we consider a user who sets an alternative ```-DCMAKE_INSTALL_PREFIX``` variable when running cmake.

It's worth considering that any user with HackRF hardware will need permission to access the device, which on most systems means having root or sudo access.  This is the same permission required to install the udev rule.  A user that installs to their home directory and who does not have sudo access will not be able to use the hardware anyway.

ckuethe, 2015-07-09:
I am one of those alternative prefix users ... /home/gnuradio. I installed the udev files for various radio by hand once, and I never need to sudo again.

users are possibly instructed to add themselves to the plugdev group, which then grants them access to the devices without root.

ckuethe, 2015-07-10:
Would it be worth not installing the rules by default, and adding an "install-rules" target? Us hackrf owners already probably have the hackrf rules in place if we need them...

ckuethe, 2015-07-15:
As a temporary method to stop the bleeding, the hackrf pybombs has been patched to not install rules by default and it's easy for users to un-patch. I've updated the FAQ to explicitly state that the pybombs recipe does not install udev rules.

ckuethe, 2015-07-16:
Also, you'll need to have Pybombs [2ade375fdd](https://github.com/gnuradio/pybombs/commit/2ade375fddb2142d92cd94d3d9986ce3d65b4a70) or newer in order for this workaround to work.

mossmann, 2017-01-26:
Thanks, ckuethe!

closed 2017-01-26 by mossmann