greatscottgadgets/hackrf #192: repeat mode (-R) incompatible with -n

mossmann, 2015-07-09:
hackrf_transfer -n (number of samples) breaks when used with -R.

I get many of these messages:
Input file end reached. Rewind to beginning.
Input file end reached. Rewind to beginning.
Input file end reached. Rewind to beginning.

dominicgs, 2015-07-09:
I think there are two ways that you would want to use -n and -R together.

1) Repeatedly transmitting the first n samples from a file.
2) Transmitting a total of n samples, repeating the file if the length of the file is less than n.

PR #194 addresses the first of these.  Is it right that the second should be an error anyway?

scateu, 2015-07-09:
for PR #194 , 
with -n -R together, when the length of file is less than n, it will repeat the file forever, and '-n' is ignored.

I think the 2) condition isn't a common usage anyway...

mossmann, 2015-07-09:
Hmm.  2 was the use case I had when I tried -R for the first time.  :-)

Example: I place a PN sequence or other short pattern in a file, and then I want to transmit that pattern for 30 seconds at a time.

For 1) you can use dd or other tools to trim the file, accomplishing the same thing.
For 2) I don't know of a better way than -n and -R together (especially with large -n).

mossmann, 2017-01-25:
I discussed with sharebrained, and we both agreed that -n should limit the total number of samples transferred over USB, not the number of samples read from a file.
Fixed in ba4ffbb.

closed 2017-01-25 by mossmann