greatscottgadgets/hackrf #193: baseband filter bandwidth sometimes not set correctly

mossmann, 2015-07-09:
Software that doesn't call hackrf_set_baseband_filter_bandwidth() when setting the sample rate sometimes gives the user a very poor baseband filter selection.  I'm thinking of fixing this by having the firmware always set a sensible filter bandwidth whenever the sample rate is changed.  Software that wants to override that selection must then call hackrf_set_baseband_filter_bandwidth() after the sample rate is set, not before.

kpreid, 2015-07-09:
FYI on precedent: `gr-osmosdr` already does this for all supported devices. And it uses `set_bandwidth(0)` to denote “please auto-select for me again”.

natevw, 2015-09-17:
Yes, I suspect this has been a cause of some long-time frustration for me with HackRF, using through GQRX. I had never put anything in its "Bandwidth" configuration as I hadn't considered what that was for or if the HackRF even supported it. So I have been plagued by a very "deaf" HackRF, which seems to be a common complaint, as well as much interference and noise problems.

Finally I discovered that for longer-band listening (with or without Ham It Up) it is veeery useful [i.e. ± *necessary*] to explicitly set a suitable bandwidth, at least to that of the sampling rate if not lower — this greatly reduces what I assume is distortion caused by strong signals outside of the sampling range but still overloading the receiver.

mossmann, 2017-01-27:
Since we already had automatic filter selection in libhackrf but not in firmware, I implemented the fix in libhackrf.  As far as I know, nobody uses HackRF without libhackrf except people who write custom firmware (e.g. PortaPack).
Fixed in f388a3f.

closed 2017-01-27 by mossmann