greatscottgadgets/hackrf #203: Patch for building 2015.07 on MSVC 2012

guruofquality, 2015-07-25:
I updated my build scripts for the 2015.07 tag. I fixed some compiler issues for MSVC2012 regarding declaring variables at the top of the function.. MSVC2013 and up support c99, but I still package for MSVC2012 which is stuck on c89. I know this it totally annoying, but the patch is small and the rest of the code seemed to be c89 compatible.

So if this is useful to upstream, the patch is here, or let me know if a pull request is preferred:
https://github.com/pothosware/PothosSDR/blob/master/patches/hackrf_fix_compat_c89_vc11.diff

dominicgs, 2016-01-26:
I have applied this patch and pushed it to master.  Thanks for the contribution, I'm sorry it took so long to merge.

closed 2016-01-26 by dominicgs

guruofquality, 2016-01-26:
Sweet! Thanks.