greatscottgadgets/hackrf #204: dfu-suffix in firmware/hackrf-common.cmake

dragonyzl, 2015-07-26:
Hi: 
In the firmware/hackrf-common.cmake,  there is a line "COMMAND dfu-suffix --vid=0x1fc9 --pid=0x000c --did=0x0 -s 0 -a _tmp.dfu".  But the option of duf-suffix is  Uppercase "S" ("-S --srec ..."). If I change it to -S,  there are some errors!  But I am not familiar with the hardware development, may someone help me? 
(The dfu-util package with version 0.8)

dominicgs, 2015-07-26:
@dragonyzl Thanks for logging this issue, it is related to a change in dfu-util between versions 0.7 and 0.8.  It is already being tracked as issue #117 (so I will close this one).

closed 2015-07-26 by dominicgs

dragonyzl, 2015-07-26:
https://github.com/dominicgs Thank you very much! I will save the https://github.com/mossmann/hackrf/issues/117 in mind for ever. Yes, it the stealth fighter F117^_^