greatscottgadgets/hackrf #205: About the latency of Hackrf

dragonyzl, 2015-07-29:
Hi, Mossmann and others:
I want to construct a transponder using two hackrf ones for some experiments.  In order to check the feasibility, I want to know the latency between the signal arrive the antenna and the digital signal write to HDD.  I think the latency should include the analog part caused by some LC circuit and the digital part caused by AD and USB. I know this could be measured by oscilloscope， but it is not at hand. Maybe someone knows the latency.  

Another question is how to control the latency, I think the latency can be controlled by adjusting the AD sample rate and the buffer size for USB reading or writting, right? Maybe someone can tell me the relationship between the latency and the sample rate and the buffer size. 

Thanks in advance!

BTW: I have implemented the some transponder function using hackrf_transfer by the FIFO file. It is very simple. 
1. Using the firmware that support the multidevice.
2. Creat a FIFO file using mkfifo in the current fold.
3. Modify the   hackrf_transfer.c like this:

result = setvbuf(fd , NULL , _IOFBF , FD_BUFFER_SIZE);
		if( result != 0 ) {
			printf("setvbuf() failed: %d\n", result);
			usage();
			return EXIT_FAILURE;
		}
==========================================================================
		// I want to modify the hackrf_transfer to implement a transponder using two hackrf one.
		// first, a FIFO file should be created, one hackrf write the received data to the FIFO.
		// and the second hackrf to read the data from the FIFO and transmits the signal. and the 
		// transponder need the some delay time.  May be the fuction setbuffer(FILE* stream, char* buf,size_t size)
		// should be used to determing the delay time by set the different buffer size.   Dragonyzl 20150728

		char mybuffer[FD_BUFFER_SIZE];
	setbuffer(fd , mybuffer ,FD_BUFFER_SIZE );  // Yes, It works. TODO: should mesure the time delay 
==========================================================================
/* Write Wav header */
	if( receive_wav ) 
	{
		fwrite(&wave_file_hdr, 1, sizeof(t_wav_file_hdr), fd);
	}

4.  One for receiver and another for transmitter by hackrf_transfer's option -r and -t with the filename of FIFO name.

Again, I want to know the time delay and its range of variation(in ns, us or ms level ). 
Thank you!

dragonyzl, 2015-07-29:
Sorry , the last "1." should be "4."

KC5CQW, 2015-08-06:
On the hardware side, it my be advantageous to apply a GPSDO (or other standard) to each HRF PLL 1 clock input to sync. Otherwise, some kind of clock correction for each HRF. 
Minimally, a null TX of (x)mS before the packet preamble to insure the RX unit has locked to the signal. 
Sorry if this is redundant...

straithe, 2021-02-14:
You are on the right track. Latency depends on a variety of factors and I encourage you to continue with your own experimentation. Have fun!

closed 2021-02-14 by straithe