greatscottgadgets/hackrf #206: can I update the cpld firmware in windows?

commthu, 2015-08-04:


dominicgs, 2015-08-04:
If you have updated both your host code and firmware to 2015.07.2, then
cpld updates should work from Windows.

Please let me know how it goes.

commthu, 2015-08-06:
I use Cmake to generate vs2012 and build the host, i have the error below:
I have pthreads-w32-2-9-1-release and libusb-1.0.18-win in the include path.
what is the possible reason? 

2>D:\HackRF\hackrf-2015.07.2\host\libhackrf\src\hackrf.c(319): error C2143: 语法错误 : 缺少“;”(在“const”的前面)
2>D:\HackRF\hackrf-2015.07.2\host\libhackrf\src\hackrf.c(320): error C2065: “libusb_error”: 未声明的标识符
2>D:\HackRF\hackrf-2015.07.2\host\libhackrf\src\hackrf.c(350): warning C4244: “=”: 从“ssize_t”转换到“int”，可能丢失数据
2>D:\HackRF\hackrf-2015.07.2\host\libhackrf\src\hackrf.c(371): warning C4244: “=”: 从“ssize_t”转换到“int”，可能丢失数据
2>D:\HackRF\hackrf-2015.07.2\host\libhackrf\src\hackrf.c(373): error C2143: 语法错误 : 缺少“;”(在“const”的前面)
2>D:\HackRF\hackrf-2015.07.2\host\libhackrf\src\hackrf.c(374): error C2065: “serial_descriptor_index”: 未声明的标识符
2>D:\HackRF\hackrf-2015.07.2\host\libhackrf\src\hackrf.c(379): error C2143: 语法错误 : 缺少“;”(在“类型”的前面)
2>D:\HackRF\hackrf-2015.07.2\host\libhackrf\src\hackrf.c(380): error C2143: 语法错误 : 缺少“;”(在“const”的前面)
2>D:\HackRF\hackrf-2015.07.2\host\libhackrf\src\hackrf.c(381): error C2065: “serial_number_length”: 未声明的标识符
2>D:\HackRF\hackrf-2015.07.2\host\libhackrf\src\hackrf.c(382): error C2065: “serial_number”: 未声明的标识符
2>D:\HackRF\hackrf-2015.07.2\host\libhackrf\src\hackrf.c(382): error C2109: 下标要求数组或指针类型

commthu, 2015-08-17:
I have updated both host and firmware to 205.07.02
hackrf_spiflash.exe  works 
but when i hackrf_cpldjtag.exe -x  hackrf_cpld_default.xsvf, get the error below:

File size 37629 bytes.
LED1/2/3 blinking means CPLD program success.
LED3/RED steady means error.
Wait message 'Write finished' or in case of LED3/RED steady, Power OFF/Disconnect the HackRF.
hackrf_cpld_write() failed: HACKRF_ERROR_LIBUSB(-1000)

jocover, 2015-08-17:
this is bug ,in line 992
result = hackrf_set_transceiver_mode(device, TRANSCEIVER_MODE_CPLD_UPDATE);
change to
result = set_hackrf_configuration(device->usb_device, TRANSCEIVER_MODE_CPLD_UPDATE);

commthu, 2015-08-17:
I change the  line 922 in the function hackrf_cpld_write

still not work and get the same error  hackrf_cpld_write failed:
HACKRF_ERROR_LIBUSB(-1000)

dominicgs, 2015-08-22:
@commthu I'm not sure what change you made to line 922, but I don't believe one is required.

Could you try this with LIBUSB_DEBUG enabled?

commthu, 2015-08-24:
how to try this with LIBUSB_DEBUG enabled?

jocover <notifications@github.com>  reply to me
in line 922
result = hackrf_set_transceiver_mode(device, TRANSCEIVER_MODE_CPLD_UPDATE);
change to
result = set_hackrf_configuration(device->usb_device, TRANSCEIVER_MODE_CPLD_UPDATE);

dominicgs, 2015-08-24:
I'm not sure how to set libusb debug mode in Windows, but in Linux you set an environment variable.  I would expect Windows to be similar.

commthu, 2015-08-24:
Can you update cpld from windows.
I am using win7.

dominicgs, 2015-08-24:
Yes, updating the cpld from Windows is possible. If you are seeing problems with the process, then I will need some more information as I have asked for in the above comments.

dominicgs, 2015-08-29:
Closing issue as CPLD updates from Windows are possible as of the 2015.07.2 release.

closed 2015-08-29 by dominicgs