greatscottgadgets/hackrf #207: Frequency Calibration and Characterization

KC5CQW, 2015-08-06:
Implement a self calibration feature to characterize the HackRF's onboard crystal reference by comparing a known standard (GPSDO, rubidium etc.) on PLL1 and the onboard temperature sensor(s). 
Establish a procedure to ramp the temperature over the expected operation temp range and store the characterization file onboard for operation without the external clock connected. This would be useful in remote/mobile operation when away from the laboratory (with time standard(I.e. PortapackH1 use etc.).
Define a realistic useful expiration limit for the calibration. (This will take a collaborative effort to collect and share data).

KC5CQW, 2015-08-06:
Perhaps the next evolutionary step is to create an open source GPSDO!
With a small ublox: 
https://shop-america.u-blox.com/en/usd/3~110~AMERICAS/Positioning-modules/Timing-raw-data/LEA-NEO-timing-modules (or equivalent),
a PLL and a high quality VCXCO, one can be incorporated into a small daughter card to connect via on board headers. Optionally, a self contained enclosure may be utilized with a trade off of an extra cable.
Note: using a freq. output vs. 1PPM will render better Allen deviation and lock time. 

***At 0.1ppm native accuracy, one of these VCXTO's from Fox:
http://www.foxonline.com/pdfs/FOX321_Standard_Specifications.pdf (or equivalent)
Would be fantastic for times when GPS signals are weak or lost.***

Secondly, the GPS will output NEMA and UTC. Some modules also contain an RTC with battery/super cap backup for warm starts. 

Reference
http://www.w1ghz.org/small_proj/A_Flexible_VCXO_Locking_Board.pdf
http://www.allanstime.com/AllanVariance/index.html

ckuethe, 2015-08-06:
Or you could just buy one...
http://www.jackson-labs.com/index.php/products/lc_xo

KC5CQW, 2015-08-06:
I could but... I already have one. Besides if more signals are needed, a simple distribution bandpass filter/buffer amp would offer many more clock signals for other equipment.
Still, a daughter card GPSDO build for the HackRF using small SMD components would yield a low power clock standard with a much smaller footprint than existing COTS units.

KC5CQW, 2015-08-21:
See also: #116

KC5CQW, 2017-12-13:
Not a significant need and TCXO’s in 0.1ppm are now very affordable.

closed 2017-12-13 by KC5CQW