greatscottgadgets/hackrf #208: hackrf_open() failed: HACKRF_ERROR_LIBUSB (-1000)

commthu, 2015-08-15:
I am using ubuntu 14.04-3
according to https://github.com/mossmann/hackrf/tree/master/host
I firstly  sudo apt-get install build-essential cmake libusb-1.0-0-dev pkg-config
and then successfully install the host
when i type hackrf_info, and get the the error:
hackrf_open() failed: HACKRF_ERROR_LIBUSB (-1000)
why?

scateu, 2015-08-15:
1. You may need to upgrade your firmware and cpld.
2. try `sudo hackrf_info`, if this works, you need to add
`-DINSTALL_UDEV_RULES=ON` to cmake command.
3. If doesn't work still, you can provide `lsusb` output for further survey.

dominicgs, 2015-08-15:
Could you try setting libusb debug mode before you run hackrf tools (export LIBUSB_DEBUG=3). That should tell us what is preventing you from using the hackrf.

dominicgs, 2015-08-22:
@commthu Were you able to try again with LIBUSB_DEBUG enabled?  Or did you manage to fix the problem?

commthu, 2015-08-24:
I have fixed the problem by re-downloading the code and re-building it.

closed 2015-08-24 by dominicgs

duchengyao, 2017-03-16:
I have the same problem and I have upgraded the `firmware 2017.02.1`, with `-DINSTALL_UDEV_RULES=ON` but still not work

https://github.com/ptrkrysik/gr-gsm/issues/269

```
root@kali:~# lsusb
Bus 002 Device 002: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 5986:066d Acer, Inc 
Bus 001 Device 004: ID 8087:0a2a Intel Corp. 
Bus 001 Device 014: ID 1d50:6089 OpenMoko, Inc. 
Bus 001 Device 003: ID 413c:301a Dell Computer Corp. 
Bus 001 Device 002: ID 046a:0011 Cherry GmbH G83 (RS 6000) Keyboard
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
root@kali:~# hackrf_info 
hackrf_info version: 2017.02.1
libhackrf version: 2017.02.1 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000909864c8283593cf
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x005e4745
```

dominicgs, 2017-03-16:
In that bug you have output that says
```
RuntimeError: hackrf_set_freq(-399053) has failed (-1000) HACKRF_ERROR_LIBUSB
```
Are you really trying to set a negative frequency?  Or is this another bug?

Could you run `find /usr -name "libhackrf*"` and paste the out put here?

duchengyao, 2017-03-17:
I have found the reason.

`grgsm_livemon -f 947.4` crashed
`grgsm_livemon -f 947400000`  and hackrf worked

Thanks!