greatscottgadgets/hackrf #209: sgpio-rx example needs updating for signed samples.

jboone, 2015-08-15:
In rx_test(), there is an incantation that converts RX samples from a buffer using "(x & 0xff) - 0x80", which is incorrect as of changes in the CPLD over a year ago. Somebody (me) was less than thorough reviewing examples...

s0pr3m3, 2015-08-17:
Tried the changes you pushed to your branch at https://github.com/jboone/hackrf/tree/sgpio_rx_samples_signed , however the example still doesn't work. sigi and sigq have value of 0. Also, shouldn’t magsq be equal to sigi * sigi + sigq * sigq instead just sigi * sigq.

jboone, 2015-08-28:
Resolved in "sgpio_rx_samples_signed" branch, pull request #211.

rroff, 2016-06-13:
Hello,

It is amazing what one can invent! You're gonna love it, I swear! Read more here <http://fophythoty.ohmygodareyouserious.com/aealkyt>

Sincerely, rroff@roff.us

closed 2017-01-27 by mossmann