greatscottgadgets/hackrf #210: HackRF Fails to Enumerate to USB Host on Boot

Dygear, 2015-08-18:
[    6.783707] usb 7-1.3: device descriptor read/64, error -71
	[   11.972805] usb 7-1.3: device descriptor read/64, error -71
	[   12.148980] usb 7-1.3: new high-speed USB device number 5 using ehci-pci
	[   17.234084] usb 7-1.3: device descriptor read/64, error -71
	[   22.423163] usb 7-1.3: device descriptor read/64, error -71
	[   22.599339] usb 7-1.3: new high-speed USB device number 6 using ehci-pci
	[   28.012542] usb 7-1.3: device not accepting address 6, error -71
	[   28.084694] usb 7-1.3: new high-speed USB device number 7 using ehci-pci
	[   33.501943] usb 7-1.3: device not accepting address 7, error -71
	[   33.502130] usb 7-1-port3: unable to enumerate USB device

This causes the boot take 30 seconds longer then it should.

The second log below shows when happens in dmesg after I press the reset button.

	[   47.764076] usb 7-1.3: new high-speed USB device number 9 using ehci-pci
	[   49.361450] usb 7-1.3: string descriptor 0 read error: -71
	[   49.361454] usb 7-1.3: New USB device found, idVendor=1d50, idProduct=6089
	[   49.361456] usb 7-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=5
	[   49.365662] usb 7-1.3: can't set config #1, error -71
	[   49.487769] usb 7-1.3: USB disconnect, device number 9
	[   49.718120] usb 7-1.3: new high-speed USB device number 10 using ehci-pci
	[   49.811175] usb 7-1.3: New USB device found, idVendor=1d50, idProduct=6089
	[   49.811180] usb 7-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=5
	[   49.811183] usb 7-1.3: Product: HackRF One
	[   49.811185] usb 7-1.3: Manufacturer: Great Scott Gadgets
	[   49.811187] usb 7-1.3: SerialNumber: 000000000000000014d463dc2f2923e1
	[   50.820604] media: Linux media interface: v0.10
	[   50.823062] Linux video capture interface: v2.00
	[   50.827668] hackrf 7-1.3:1.0: Board ID: 02
	[   50.827670] hackrf 7-1.3:1.0: Firmware version: 2015.07.2\xffffffdf^
	[   50.828109] hackrf 7-1.3:1.0: Registered as swradio0
	[   50.828111] hackrf 7-1.3:1.0: SDR API is still slightly experimental and functionality changes may follow
	[   50.828124] usbcore: registered new interface driver hackrf

I'm running Ubuntu 15.04 with all the patches.

dominicgs, 2015-08-18:
This is strange, I would expect the process at boot time to be identical to the process when the device is plugged in / reset.

Once you have reset the device, does it work as expected?

Could you describe the host hardware that you are using?  Which version of HackRF hardware is this?

I see that you are running the latest firmware, although that version string looks bad, perhaps the firmware flashing had errors?  Could you try to re-flash it?  Failing that, could you attempt a downgrade to the previous release firmware?

Dygear, 2015-08-18:
For when this error happens, I have tested it on two different host computers. I've made a Live USB drive where I installed Ubuntu 15.04, I connected this drive to the USB 3 ports on my computer with no other USB device attached. The first computer is as follows.

Common Equipment:
Drive: [SanDisk Extreme CZ80 64GB USB 3.0 Flash Drive](http://www.amazon.com/gp/product/B00KT7DOSE?psc=1&redirect=true&ref_=oh_aui_detailpage_o02_s00)

Desktop: (Built Myself)
Intel® Core™ i7-3770K CPU @ 3.50GHz x 8
16GB of Corsair RAM DDR3-3200 (MemChecked, All Good)

Laptop: (Apple Retina Macbook Pro 13")
Intel® Core™ i5-3210M CPU @ 2.50GHz × 4
8GB of Apple RAM DDR3-3200 (MemChecked, All Good)

For the laptop, both USB 3 ports are directly connected to the Processor, so they should offer the full bandwidth. For the desktop, I made sure to connect it to the ports that are closest too the CPU according to the motherboards information (ASUS P8Z77-V DELUXE LGA 1155).

I'm using the production version of HackRF.

I had to flash the device multiple times, including flashing from DFU mode due to the first flash failing. After the upgrade, it works but with a degradation performance over the 2014 firmware version that it had. Not really sure why as of yet, I assumed that it was just par for the course. I'll give a downgrade a shot to the firmware that it came with and get back to you.

Dygear, 2015-08-19:
No issues with the 2015.07.1 version, let's see if I can reproduce it with the 2015.07.1

	[    1.690882] usb 7-1.3: new high-speed USB device number 4 using ehci-pci
	[✂️]
	[    1.784040] usb 7-1.3: New USB device found, idVendor=1d50, idProduct=6089
	[    1.784042] usb 7-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=5
	[    1.784043] usb 7-1.3: Product: HackRF One
	[    1.784044] usb 7-1.3: Manufacturer: Great Scott Gadgets
	[    1.784045] usb 7-1.3: SerialNumber: 000000000000000014d463dc2f2923e1
	[✂️]
	[    4.875317] hackrf 7-1.3:1.0: Board ID: 02
	[    4.875320] hackrf 7-1.3:1.0: Firmware version: 2015.07.1
	[    4.875743] hackrf 7-1.3:1.0: Registered as swradio0
	[    4.875744] hackrf 7-1.3:1.0: SDR API is still slightly experimental and functionality changes may follow
	[    4.875775] usbcore: registered new interface driver hackrf

dominicgs, 2015-08-19:
As far as I remember, the differences between 2015.07.1 and 2015.07.2 were only in host code, so I'm hopeful that 2015.07.2 will work just as well as 2015.07.2.  If that is the case, I will put this down to the firmware write failing in some way.

dominicgs, 2015-08-22:
@Dygear Were you able to reproduce the problem with 2015.07.2 firmware?

dominicgs, 2015-08-29:
@Dygear Are you able to confirm that the issue was resolved by re-flashing the firmware?

I will close this issue in a few days if there are no further problems.

closed 2015-12-08 by Dygear