greatscottgadgets/hackrf #216: Add Calibration

Gamephreaker, 2015-08-29:
Mine is off by more then 5 Mhz really hard to use I have hunt for broadcasts

dominicgs, 2015-08-29:
5MHz sees like a very big offset and is more likely to be related to other issues than to a drifting clock.  Which hardware version are you using?  Which versions of firmware, host tools and CPLD firmware are you using?

Adding calibration functions is already being tracked as #207, although there are some limitations to the effectiveness of calibration techniques and their results over time.

Gamephreaker, 2015-08-29:
I am using a hackrf one. I have used the current builds and now the portapack build. It does it on the portapack as well.

natevw, 2015-09-17:
@Gamephreaker Are you perhaps picking up harmonics of strong signals instead? What software and filter/gain settings are you using?

mossmann, 2017-01-27:
Closing due to lack of response from original poster.  This was almost certainly either a bug or a user error, not a calibration issue.  If further assistance is needed, please see: [https://github.com/mossmann/hackrf/wiki/Getting-Help](https://github.com/mossmann/hackrf/wiki/Getting-Help)

closed 2017-01-27 by mossmann