greatscottgadgets/hackrf #217: Add USB-initiated reboot vendor request

jboone, 2015-09-07:
It'd be handy to simplify firmware re-flashing, if after flashing, the HackRF could be asked to restart with the new firmware.

I'm sure there's security implications around having a mechanism to download new firmware and then execute it. (Oh hai, mass storage device!) But it would make things a bit easier for the user... I have code in the PortaPack project that accomplishes this. If it's deemed an OK idea, I can proceed with building a pull request.

dominicgs, 2015-09-07:
I think this is a reasonable feature to expect the firmware to have built in.  Some users run HackRF in an environment where pressing the reset button is not trivial, so resetting from software would be handy.

I understand your concerns, but DFU allows code to be written and run immediately, so I think it's reasonable to expect the same from our firmware update mechanism.

If you have the code ready to go, then I'd say that you should go ahead and submit a pull request.

dominicgs, 2017-01-27:
Software reset is added in PR #313

closed 2017-01-27 by dominicgs