greatscottgadgets/hackrf #219: UDEV_INSTALL_DIR is ignored by cmake building 2015.07.02

barjac, 2015-09-09:
cmake ignores  UDEV_INSTALL_DIR

/etc/udev/rules.d seems to be hard coded and we have had to patch the CMakeList.txt in previous versions for this issue. We need to use /usr/lib/udev/rules.d
Grepping the source:
[baz@localhost hackrf-2015.07.2]$ grep -r '\/etc\/udev\/rules\.d'
host/libhackrf/CMakeLists.txt:    "/etc/udev/rules.d"
build/libhackrf/cmake_install.cmake:   "/etc/udev/rules.d/53-hackrf.rules")
build/libhackrf/cmake_install.cmake:file(INSTALL DESTINATION "/etc/udev/rules.d" TYPE FILE FILES "/home/baz/BLD/BLD_SDR_STACK/hackrf_2015/BUILD/hackrf-2015.07.2/build/libhackrf/53-hackrf.rules")
build/CMakeCache.txt:UDEV_RULES_PATH:STRING=/etc/udev/rules.d
build/install_manifest.txt:/etc/udev/rules.d/53-hackrf.rules

It does look as though the new value for UDEV_INSTALL_DIR has been picked up - it's just not used :\

[baz@localhost hackrf-2015.07.2]$ grep -r UDEV_INSTALL_DIR
build/CMakeCache.txt:UDEV_INSTALL_DIR:UNINITIALIZED=/usr/lib/udev/rules.d
[baz@localhost hackrf-2015.07.2]$

dominicgs, 2015-09-11:
Thank you for reporting this.  This is related to a known issue when installing as a non-root user as I believed that there was nowhere else that installing udev rules made sense, other than /etc/udev/rules.d

I will fix this so that UDEV_INSTALL_DIR is obeyed.

dominicgs, 2017-01-27:
The path of the installed udev rule can be controlled by passing `-DUDEV_RULES_PATH=/path/to/udev` to cmake.  I have updated the host/README.md file to represent this.

closed 2017-01-27 by dominicgs