greatscottgadgets/hackrf #221: Enhancement request: provide SoapySDR module for HackRF

dc1rdb, 2015-09-22:
This would allow easy integration of HackRF into any SDR software that uses the SoapySDR library. Details can be found at https://github.com/pothosware/SoapySDR/wiki

dominicgs, 2015-09-23:
Thanks for the suggestion.  HackRF is currently supported by SoapySDR as part of the OsmoSDR module.  I'll leave this issue open so that we can discuss the pros and cons of writing a separate HackRF module.

guruofquality, 2015-09-28:
Its up: https://github.com/pothosware/SoapyHackRF/wiki

dominicgs, 2015-09-28:
@guruofquality That's great!  Thank you.

Were there any issues/stumbling blocks?  Anything we should modify in libHackRF?

guruofquality, 2015-09-30:
For the most part everything went smoothly.

One of the original issues that we ran into w/ SoapyOsmo was instantiating the tx and rx streams at the same time resulted in an error. I understand the RF frontend is half duplex, but I'm not positive if we can or cant have two active data streams.

I would like to make it possible for both a hackrf source and sink block to coexist in the same flow graph for the same device. At the moment, this means using the activation/deactivation hooks in soapy sdr to start up and stop the respective streams. And so with a little creative control, a user could stop rx and start tx, send a burst, and switch back. We could also try to make this switching somehow implicit in the soapy sdr wrapper when the user transmits -- using an end of burst flag to restart the rx streaming -- and skipping the whole activation/deactivation hooks I mentioned.

I can potentially support all of that above with the existing hackrf api as I understand it. My concern is have I misunderstood something about the simultaneous streams, the cost of starting and stopping streams vs something more efficient, and is there anything in libhackrf that could be done to facilitate something like that.

I hope that makes sense. And regardless of how its implemented, I'm curious if this sounds like a worthwhile "feature" to implement in SoapyHackRF. Thanks!

mossmann, 2017-01-25:
Thanks, guruofquality!

closed 2017-01-25 by mossmann