greatscottgadgets/hackrf #223: GPS L1 band with HackRF

mouradghafiri, 2015-11-04:
Hello;

First of all I would like to express my deep gratitude to the team behind this great device.
Well, I'm working on GPS L1 positioning using software approach, I don't have enough knowledge in the hardware side of this implementation, all my work is focusing on the software side of the positioning process, so I would like to have a streaming that contains the information of the L1 band.

I know it is possible with HackRF, so please would help me to have that streaming using GNU Radio... if there is an already implemented gnu radio project that show that spectrum of L1 Band and save it to a file for post processing, as well as the specification for the antenna used. I will really appreciate that.

Thank you so much in advance.

scateu, 2015-11-04:
github.com/osqzss/gps-sdr-sim

mouradghafiri, 2015-11-04:
Thank you so much for this quick answer,
Actually the project you have mentioned just simulate the GPS signals, for my case i'm using real signals (signals that contains all sorts of errors and distortions).

So all I need is to have that real streaming with a specific rate and IF (in case it is used) , so I can go further with the acquisition and tracking process then the PVT calculation.

You may want to check this link to get the point 
https://www.youtube.com/watch?v=TWXBvv9opCA

Regards;

scateu, 2015-11-04:
Actually, the generated signal samples can be feeded directly in to 
HackRF. I have done a few of experiments. Both serial-based GPS receiver 
and smartphones can decode this simulated signal through antenna.

As for realtime streaming, you may try UNIX pipe, set the output of 
gps-sdr-sim to stdout, then set the input of hackrf_transfer to stdin.

For example ( I haven't tried, but it probably works.):

     $ mkfifo gps.fifo
     $ gps-sdr-sim -e brdc3540.14n -l 30.286502,120.032669,100 -o gps.fifo 
&
     $ hackrf_transfer -t gps.fifo -f 1575420000 -s 2600000 -a 1 -x 0

--
Wang Kang
Fingerprint: 011F 0492 97D6 5D75 8AC4  6458 D43F 3CE2 3353 B7BD

mouradghafiri, 2015-11-04:
Dear scateu,
I really appreciate your help, but what you are talking is transmission of simulated GPS signal through HackRF. In my case it is the inverse... I want just to capture the L1 band streaming using HackRF.
So it is all about a small project with osmocom source block and FFT Spectrum block and a File sink. the appropriate configuration of these blocks is what I miss in my implementation as well as the specification of the antenna.

Best Regards;

dominicgs, 2015-11-04:
@mouradghafiri The version in the video that you linked will work with HackRF if you change the source block.

There are also many GPS/GNSS SDR implementations that don't rely on GNU Radio, some are listed here: https://en.wikipedia.org/wiki/GNSS_software-defined_receiver

mouradghafiri, 2015-11-04:
@dominicgs
I have already played with the configuration of the source block... no significant results as in the video... I have sent an email to the owner of that video asking him for the specification of the antenna used and the configuration of the block used... but unfortunately, no feedback.

Hope you can help

dominicgs, 2015-11-04:
I'm not sure that I can help with the information provided.  Maybe you could share the GNU Radio companion flowgraph that you've been trying to use?

dominicgs, 2015-11-04:
Additionally, these sort of queries are often better dealt with via the HackRF mailing list: https://pairlist9.pair.net/mailman/listinfo/hackrf-dev

The GitHub issue tracker is normally used for feature requests or bugs in the HackRF software/firmware.

mouradghafiri, 2015-11-04:
Ok got it, I will share the GNU Radio companion flowgraph in the mailing list.
Thank you so much

closed 2016-01-14 by dominicgs