greatscottgadgets/hackrf #225: Hackrf One over ip

aler86, 2015-11-17:
Hello,

I have bougth HackRF one and tested it with FM,GSM,DVB in gnuradio.

Now i'm tried hackrf one with IEEE 802.11 a/g and have a question:

Is it possible assign ip address to Hackrf one maybe through a serial-ethernet converter?

Thanks in advance

aler86, 2015-11-17:
sorry....i'm trying hackrf one with IEEE 802.11 a/g and...

dominicgs, 2015-11-17:
When it comes to network protocols, such as 802.11a/g, the HackRF hardware is the radio front-end.  The IP address or MAC address would be set in the software implementation of 802.11

aler86, 2015-12-10:
Can i use tun/tap for Hackrf one?

dominicgs, 2015-12-14:
There isn't any existing software to use tun/tap with HackRF.

What is it that you are trying to achieve?

aler86, 2015-12-14:
I'm trying to achieve 802.11 a/g protocol

dominicgs, 2015-12-17:
HackRF One is just the radio, everything else will need to be written in software.  If you wish to pass packets to your 802.11 stack via tun/tap, then you will need that to be built in to the software stack.

closed 2016-01-14 by dominicgs