greatscottgadgets/hackrf #226: How to reboot the device without touch the device?

commthu, 2015-11-18:
sometime, the device is dead, I dont know the reason. 
I have to disconnect the usb cable and connect it again.
But if the device is in the remote place,  How to reboot the device?

dominicgs, 2015-11-18:
There isn't currently a software reset function, although we've discussed adding one.

If you need to reset the device there is no need to disconnect the USB cable, you can simply press the reset button on HackRF One.

commthu, 2015-11-19:
I am expecting you to add the software reset the device.

ckuethe, 2015-11-24:
If the device is hung, how is a program going to be able to reset it?

You perhaps you can get linux to reset the device. Here are some things I found when I googled for "linux usb reset"
http://billauer.co.il/blog/2013/02/usb-reset-ehci-uhci-linux/
http://www.linux.org/threads/resetting-the-usb-subsystem.6256/
http://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line

commthu, 2015-11-25:
thank you!
But we are using the hackrf in windows.

dominicgs, 2015-12-14:
@commthu The question still remains: if the device is not responding, how will adding a reset function to the firmware help?

I will leave this issue open as there may be times that a reset function is useful, but I think finding a way to force Windows to reset the device would better solve your issue.

dominicgs, 2016-01-26:
Closing as a duplicate of #217

closed 2016-01-26 by dominicgs

rroff, 2016-06-13:
Hi,

I've got some new pizza recipe for you, you should try it, more info here <http://curduhyche.bitingthehandthatfeeds.com/aeapisj>

rroff@roff.us

rroff, 2016-06-13:
Hi,

Have you seen that stuff? It's something crazy, just take a look <http://dumandone.theweddingdancespecialists.net/aebqijbv>

Looking forward, rroff@roff.us