greatscottgadgets/hackrf #227: HackRF One not found and USB LED no light after firmware upgrade

ran2, 2015-11-29:
After executed: hackrf_spiflash -w hackrf_jawbreaker_usb_rom_to_ram.bin 
Result: HackRF is no longer responsive (e.g. hackrf_info reports no HackRF boards found)
I tried: creating proposed 52-hackrf.rules file at /etc/udev/rules.d, then reboot machine. Issue still not resolved.

dominicgs, 2015-12-01:
It looks like you have flashed the jawbreaker firmware on to a HackRF One and now it's coming up in NXP's DFU mode.  Don't worry, this is easily fixed.

First you need to plug the HackRF One in and then hold down the "DFU" button while you press the "reset" button.  This will make sure that it is in DFU mode.  You can check by looking at dmesg or lsusb, you should see similar output to the output that you listed above.

Next, you need to install `dfu-util`, if you're using a Debian based distro, this should be `apt-get install dfu-util`.  For other distros you will need to find it from your package manager.

Change to the downloaded release directory and write the temporary RAM only firmware using:
`dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb_ram.dfu`

Once that has finished, you should see an "OpenMoko" device in the output from lsusb.  You should see a Great Scott Gadgets HackRF listed in dmesg (this is due to aspect of how USB is managed)

Finally, you can write the correct firmware using: 
`hackrf_spiflash -w hackrf_one_usb_rom_to_ram.bin`

After resetting the hardware you should see the OpenMoko device in lsusb and the Great Scott Gadgets HackRF listed in dmesg output.

Please let us know if this doesn't work, or close the issue if it resolves the problem for you.

ran2, 2015-12-02:
Many thanks Dominic, but it is not working...Here what I've got:

~/sdr/hackrf/hackrf-2015.07.2/firmware-bin$ dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb_ram.dfu

dfu-util 0.5

dfu-util does currently only support DFU version 1.0

Filter on vendor = 0x1fc9 product = 0x000c
Opening DFU USB device... ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Runtime Interface...
Determining device status: state = dfuIDLE, status = 0
WARNING: Runtime device already in DFU state ?!?
Found Runtime: [1fc9:000c] devnum=0, cfg=1, intf=0, alt=0, name="DFU"
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
DFU CRC does not match
Unsupported DFU file revision 0000

I tried to upload the current DFU file. It's size is 131072 with md5 hash of 0dfbe8aa4c20b52e1b8bf3cb6cbdf193.

I don't found from the dfu-util to force write hackrf_one_usb_ram.dfu.  Any advices?

dominicgs, 2015-12-02:
Dfu util 0.5 is rather old; we recommend 0.7. Which distribution are you
using?

ran2, 2015-12-03:
After installing dfu-util 0.8, problem solved....Many thanks Dominic. Learned more on researching the "Device Firmware Upgrade Protocol" because of this mistakes....Cool

dominicgs, 2015-12-03:
Great glad we could help.  I'll close this issue now as it is resolved.

closed 2015-12-03 by dominicgs

aga06tr, 2025-06-20:
[image]

Is there anyone who can help?

aga06tr, 2025-06-20:
https://github.com/user-attachments/assets/0596515c-25e6-4197-8248-3d8fa67b1764

aga06tr, 2025-06-20:
Windows does not see the USB at all.