greatscottgadgets/hackrf #230: Add the crystal error (ppm) correct function to the hackrf_transfer

uhwsin, 2016-01-12:
Intro:
Some of the Hackrf board use a bad performance crystal, it will cause some program can't work---such as GNSS(GPS) simulation.  Most GPS receiver can capture the satellite
when the doppler within 10KHz, but 10ppm bias of the crystal will cause  15Khz doppler.  And my hackrf board uses a 16ppm crystal :(  normal
Of course, I can replace the crystal or use a external clock. But for most people, add a ppm correct function to hackrf_transfer is useful.
How to use:
1, measure your crystal's error
You can use a precision  counter to measure it.
Connect the hackrf's "clock out" to the counter, get the frequency Fh, then ,
ppm = (Fh - 10000000)/10 
 
You can use a standard clock source such as OCXO, TCXO, or Atom Clock, and a counter
to measure the crystal error too.

2, add an argument "-C ppm " when you use hackrf_transfer.

Eg. , your  crystal error is -8ppm ( In other words , your clock is slower than the standard clock), then you should add "-C -8" follow the hackrf_transfer command.

Hope you can understand my poor English:)

--------------------------
Thanks to Michael Ossmann;
Thanks to Jared Boone ;
Thanks to Benjamin Vernoux ;
Thanks to GPL.
[Uploading modified_hackrf_transfer.src.zip…]()
[Uploading modified_hackrf_transfer.src.zip…]()

uhwsin, 2016-01-12:
upload the src file again

[Uploading modified_hackrf_transfer.src.zip…]()

uhwsin, 2016-01-12:
again
[modified_hackrf_transfer.src.zip](https://github.com/mossmann/hackrf/files/87308/modified_hackrf_transfer.src.zip)

dominicgs, 2016-01-13:
@uhwsin Thank you for the code.  We usually prefer that contributions are made using the GitHub pull request system.  This allows us to view and manage the code in a transparent way that helps people who are looking at the code in the future.

Would it be possible for you to submit this change as a pull request from your repository?

uhwsin, 2016-01-13:
@dominicgs Thank you! I'll make a pull then delete this issue .

closed 2016-01-13 by uhwsin