greatscottgadgets/hackrf #231: PDF downloads from repository corrupt

sbmocp, 2016-01-12:
When trying to download any of the PDFs from the repository I get a message stating the download is corrupt or not a valid PDF. I've tried both the latest Adobe PDF reader and FoxIt with the same results. Using Windows 7 64-bit on my PC.

dominicgs, 2016-01-12:
Do you mean the PDFs from the repository?  Are you downloading them from the GitHub web interface or checking out the repository?

Does this link work for you? https://github.com/mossmann/hackrf/raw/master/doc/LPCXPresso_Flash_Debug_Tutorial.pdf

lukeweston, 2016-01-12:
This isn't a bug, more like a "how do I use the GitHub web interface"
question.

FYI, for beginners, say you're at the following page, for example.
https://github.com/mossmann/hackrf/tree/master/doc

Suppose LPC4330_SGPIO_SignalGenerator_Measurements.pdf is the file you
want, so click on that link.

Now click on the little "Raw" box on the right side, and the PDF file will
download/open for you.

sbmocp, 2016-01-12:
lukeweston had it. Many thanks--it didn't work the way I'd expected.

closed 2016-01-12 by sbmocp