greatscottgadgets/hackrf #235: Problems compiling HackRF Tools

GermMuz, 2016-01-24:
I have followed the instructions for installing and setting up the environment for compiling the hackrf tools using Visual Studio 2012. Cmake successfully creates the .sln files. I have tried this on win 7 and win 10 and get multiple compile errors along the lines of:

syntax error: missing ';' before 'const'

there also seem to be  an int to bool issue on line 32

I would really appreciate some help here as I have no idea what to do now.

Is it better to use a different compiler????

I have tried zefies tools but these do not work, giving an error 5, device not found. SDR Sharp works perfectly, but I need a hackrf.dll without copyright issues for adding to one of my programs.

Thx

dominicgs, 2016-01-24:
I haven't built the HackRF tools on a Windows host before, but I have heard of people having success with the Visual Studio compiler.  Could you copy and paste the entire build log so that we can see where these errors are coming from?

Also, which version of the HackRF code are you using, the latest release or git?

We don't ship binary files, so there is no official hackrf.dll file available.  I'm not sure I know of a reason that the version shipped with SDR# would be problematic, assuming that it is just a compiled copy libhackrf.

GermMuz, 2016-01-25:
Firstly the output if running cmake

C:\>cd C:\hackrf-2015.07.2\host\cmake

C:\hackrf-2015.07.2\host\cmake>cmake ../ -G "Visual Studio 11 2012 Win64" -DLIBUSB_INCLUDE_DIR=c:\libusb-1.0.18-win\include\libusb-1.0 -DLIBUSB_LIBRARIES=c:\libusb-1.0.18-win\MS64\static\libusb-1.0.lib -DTHREADS_PTHREADS_INCLUDE_DIR=c:\pthreads-w32-2-9-1-release\Pre-built.2\include -DTHREADS_PTHREADS_WIN32_LIBRARY=c:\pthreads-w32-2-9-1-release\Pre-built.2\lib\x64\pthreadVC2.lib
-- The C compiler identification is MSVC 17.0.50727.1
-- The CXX compiler identification is MSVC 17.0.50727.1
-- Check for working C compiler using: Visual Studio 11 2012 Win64
-- Check for working C compiler using: Visual Studio 11 2012 Win64 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler using: Visual Studio 11 2012 Win64
-- Check for working CXX compiler using: Visual Studio 11 2012 Win64 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Threads: c:/pthreads-w32-2-9-1-release/Pre-built.2/lib/x64/pthreadVC2.lib
-- Configuring done
-- Generating done
-- Build files have been written to: C:/hackrf-2015.07.2/host/cmake

C:\hackrf-2015.07.2\host\cmake>

GermMuz, 2016-01-25:
Now the compile using visual studio 2012

Error	1	error C2143: syntax error : missing ';' before 'const'	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	319	1	hackrf
Error	2	error C2143: syntax error : missing ';' before 'const'	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	319	1	hackrf-static
Error	3	error C2065: 'libusb_error' : undeclared identifier	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	320	1	hackrf-static
Error	4	error C2065: 'libusb_error' : undeclared identifier	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	320	1	hackrf
Warning	5	warning C4244: '=' : conversion from 'ssize_t' to 'int', possible loss of data	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	350	1	hackrf
Warning	6	warning C4244: '=' : conversion from 'ssize_t' to 'int', possible loss of data	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	350	1	hackrf-static
Warning	7	warning C4244: '=' : conversion from 'ssize_t' to 'int', possible loss of data	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	371	1	hackrf-static
Error	8	error C2143: syntax error : missing ';' before 'const'	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	373	1	hackrf-static
Error	9	error C2065: 'serial_descriptor_index' : undeclared identifier	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	374	1	hackrf-static
Error	10	error C2143: syntax error : missing ';' before 'type'	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	379	1	hackrf-static
Error	11	error C2143: syntax error : missing ';' before 'const'	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	380	1	hackrf-static
Error	12	error C2065: 'serial_number_length' : undeclared identifier	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	381	1	hackrf-static
Error	13	error C2065: 'serial_number' : undeclared identifier	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	382	1	hackrf-static
Error	14	error C2109: subscript requires array or pointer type	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	382	1	hackrf-static
Error	15	error C2065: 'serial_number' : undeclared identifier	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	383	1	hackrf-static
Warning	16	warning C4047: 'function' : 'const char *' differs in levels of indirection from 'int'	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	383	1	hackrf-static
Warning	17	warning C4024: 'strdup' : different types for formal and actual parameter 1	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	383	1	hackrf-static
Warning	18	warning C4244: '=' : conversion from 'ssize_t' to 'int', possible loss of data	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	371	1	hackrf
Error	19	error C2143: syntax error : missing ';' before 'const'	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	373	1	hackrf
Error	20	error C2065: 'serial_descriptor_index' : undeclared identifier	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	374	1	hackrf
Error	21	error C2143: syntax error : missing ';' before 'type'	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	379	1	hackrf
Error	22	error C2143: syntax error : missing ';' before 'const'	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	380	1	hackrf
Error	23	error C2065: 'serial_number_length' : undeclared identifier	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	381	1	hackrf
Error	24	error C2065: 'serial_number' : undeclared identifier	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	382	1	hackrf
Error	25	error C2109: subscript requires array or pointer type	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	382	1	hackrf
Error	26	error C2065: 'serial_number' : undeclared identifier	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	383	1	hackrf
Warning	27	warning C4047: 'function' : 'const char *' differs in levels of indirection from 'int'	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	383	1	hackrf
Warning	28	warning C4024: 'strdup' : different types for formal and actual parameter 1	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	383	1	hackrf
Warning	29	warning C4267: '=' : conversion from 'size_t' to 'int', possible loss of data	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	427	1	hackrf
Error	30	error C2143: syntax error : missing ';' before 'type'	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	449	1	hackrf
Error	31	error C2143: syntax error : missing ';' before 'const'	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	450	1	hackrf
Error	32	error C2065: 'serial_number_length' : undeclared identifier	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	451	1	hackrf
Error	33	error C2065: 'serial_number' : undeclared identifier	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	452	1	hackrf
Error	34	error C2109: subscript requires array or pointer type	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	452	1	hackrf
Error	35	error C2065: 'serial_number' : undeclared identifier	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	453	1	hackrf
Error	36	error C2065: 'serial_number' : undeclared identifier	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	454	1	hackrf
Warning	37	warning C4047: 'function' : 'const char *' differs in levels of indirection from 'int'	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	454	1	hackrf
Warning	38	warning C4024: 'strncmp' : different types for formal and actual parameter 1	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	454	1	hackrf
Error	39	error C2065: 'serial_number_length' : undeclared identifier	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	463	1	hackrf
Error	40	error C2143: syntax error : missing ';' before 'type'	C:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	602	1	hackrf
[... cut ...]
	87	IntelliSense: argument of type "void *" is incompatible with parameter of type "libusb_device *"	c:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	375	22	hackrf-static
	88	IntelliSense: argument of type "void *" is incompatible with parameter of type "libusb_device *"	c:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	604	18	hackrf-static
	89	IntelliSense: argument of type "transceiver_mode_t" is incompatible with parameter of type "hackrf_transceiver_mode"	c:\hackrf-2015.07.2\host\libhackrf\src\hackrf.c	922	47	hackrf-static

jboone, 2016-01-26:
I do not have the same compiler version, so I can't reproduce. However, reviewing the source code where the errors occur, these errors are pretty clearly from a C89 compiler attempting to compile a C99 or newer codebase, where variable declarations can appear after the first statement in a block. I believe C99 support came in VS 2013, and indeed, my VS 2013 compiler does not complain about those parts of the code (I have linker problems, but the compilation is successful and clean enough).

dominicgs, 2016-01-26:
I have applied a patch to the git master branch that should fix the build errors caused by MSVC's C89 rules.

@GermMuz Until the next release you have two options:
1) Upgrade to a newer version of the compiler - the free community version is supported
2) Use the git codebase rather than the release

I recommend option 2.

GermMuz, 2016-01-26:
Help appreciated, I'll try the latest build in a minute. I did try extracting the source code and making my own dll only having coded a few functions and that worked, I'll report back later.

GermMuz, 2016-01-26:
I've now compiled the code (x64 version) and all is working well. 

Thank you

closed 2016-01-26 by GermMuz

GermMuz, 2016-01-26:
Thank you for the reply.

 

The problem is now fixed and I have the code compiled. 

 

I did try vs2015 first and run into the identical compile problems, then got out the old vs2012 disks.

 

Thank you for the fix, really appreciated.

 

Glenn