greatscottgadgets/hackrf #237: Couldn't transfer any bytes for one second.

nneil, 2016-01-29:
I am using `hackrf_transfer` to capture data at 20MS/s, which equates to about 40MB/s of data to write to disk.

Tests with `dd` show I can sustain about 78.0 MB/s to disk.

If I write to `/dev/null` then `hackrf_transfer` is fine, but if I try writing to a file it randomly stops after a while with a message like this:
```
40.1 MiB / 1.000 sec = 40.1 MiB/second
40.1 MiB / 1.000 sec = 40.1 MiB/second
39.8 MiB / 1.000 sec = 39.8 MiB/second
40.1 MiB / 1.000 sec = 40.1 MiB/second
39.8 MiB / 1.000 sec = 39.8 MiB/second
40.1 MiB / 1.000 sec = 40.1 MiB/second
40.1 MiB / 1.000 sec = 40.1 MiB/second
39.8 MiB / 1.000 sec = 39.8 MiB/second
12.3 MiB / 1.000 sec = 12.3 MiB/second
 0.0 MiB / 1.000 sec =  0.0 MiB/second

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 133.01428 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
```
In the example above it managed 4.8G of data. I have tried using a separate disk just for the captured data.

dominicgs, 2016-01-29:
```
In the example above it managed 4.8G of data.
```
Does this number vary?  And while it's related it's worth asking, does the time vary too?

I think it's unlikely to be anything to do with the HackRF hardware/firmware if it works when writing to /dev/null, so we should probably focus on the HackRF tools and other host code.

Could you tell me which version of the HackRF tools you have installed?

nneil, 2016-01-29:
It's essentially random as to how long it works for. The data captured is of course proportional to the amount of time it runs. Even at 10 Ms/sec the problem can occur. I am also saddened by the fact that the transfer does not manage a consistent rate. It might drop from 20 to 12 MS/s: this must mess up my results.

I just downloaded the git version, having tried another version. I have tried two different PCs one a quad core desktop and one a quad core laptop.

How do I get `hackrf_transfer` to tell me its version number?

This is at 10MS/s and produced 1.5G of data:

```
20.2 MiB / 1.000 sec = 20.2 MiB/second
19.9 MiB / 1.000 sec = 19.9 MiB/second
18.1 MiB / 1.000 sec = 18.1 MiB/second
 0.0 MiB / 1.000 sec =  0.0 MiB/second

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 82.00829 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit

```

dominicgs, 2016-01-29:
hackrf_transfer doesn't have a version number option, so just knowing the version of the host tools that you installed is probably enough.  How did you install them?  I'd expect it to be either release 2015.07 or from git?

I can't think of anything else that would cause this, especially as it's fine when piping to /dev/null.  Do you use other tools with HackRF, such as gqrx or GNU Radio?  Do they suffer from this issue?

The data rate temporarily dropping by a small amount won't affect your data because the sample rate of the radio doesn't change and the slack is taken up by the buffer on the HackRF, which often recovers later. If the transfer rate drops too much, the buffers fill up and the whole thing becomes unsustainable.  This may be what's happening here, but I would expect it to happen no matter where you're writing data to.

nneil, 2016-01-29:
Does it work fine for you? Can you try to reproduce the problem? I think it is a `hackrf_transfer` bug.

dominicgs, 2016-01-29:
I've tried to reproduce the problem, but so far I don't see any issues writing to disk for prolonged periods of time.

nneil, 2016-01-30:
What OS are you using?

dominicgs, 2016-01-30:
Ubuntu 15.10, kernel 4.2.0

nneil, 2016-02-01:
I set up Ubuntu on the same system and the problem is the same. It's much worse if I use a longer USB cable, which is worrying, but it's is still an issue with the standard cable. I've tried USB 3 and USB 2 ports on the desktop, though my laptop only has USB 3. I will borrow another hackrf to try.

dominicgs, 2016-02-01:
Could you try mounting a ramdisk and writing to it to see if that suffers from the same issue?

Right now I can only think of a few things that would cause this, most of which would be limited to your environment, but that's becoming less and less likely as you try more and more environments.

The issue changing when you switch cables is an interesting data point, but I can't think of why that would happen.  I'm especially confused by the issue not presenting itself when you write to /dev/null

I'm racking my brain for ideas of what could be going wrong within your HackRF, but my only thought is something wrong with the USB connection/controller on the board, but I would expect that to present itself more often.

nneil, 2016-02-02:
I can confirm that `/dev/null` is fine. It runs until I stop it:
```
39.8 MiB / 1.000 sec = 39.8 MiB/second
40.1 MiB / 1.000 sec = 40.1 MiB/second
40.1 MiB / 1.000 sec = 40.1 MiB/second
39.8 MiB / 1.000 sec = 39.8 MiB/second
^CCaught signal 2
11.8 MiB / 0.296 sec = 39.9 MiB/second

User cancel, exiting...
Total time: 6302.94092 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
```
The ramdisk is also fine: it runs until the 6G I allocated fills up, though it reports this as **User cancel**
```
39.8 MiB / 1.000 sec = 39.8 MiB/second
40.1 MiB / 1.000 sec = 40.1 MiB/second
40.1 MiB / 1.000 sec = 40.1 MiB/second
39.8 MiB / 1.000 sec = 39.8 MiB/second
28.6 MiB / 1.000 sec = 28.6 MiB/second
39.6 MiB / 1.000 sec = 39.6 MiB/second

User cancel, exiting...
Total time: 162.01701 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
```
I tested again with a long cable. But can't reproduce the terrible problems I had. So I guess we we should ignore that data point.

dominicgs, 2016-02-02:
It seems that this is only happening when writing to disk, not for general data transfer to the host.  I think that it may be something to do with disk write caching / filesystem driver / etc, but that's not an area that I know much about.

dominicgs, 2016-02-19:
@nneil Were you able to borrow another HackRF to test with your system?  I still believe that this is going to come down to disk performance or at least some interaction between the disk and USB controllers.

nneil, 2016-02-19:
I did borrow another HackRF, but haven't had enough time to prove anything other than the problem is there for both of them. I don't understand why you can't reproduce it! I seem to be very good at having problems no-one else has. I think it's poor disk handling by the kernel. My Centos server seems to be more consistant at sustaining high disk data rates than Ubuntu or Fedora workstation. The Centos system is normal PC hardware, nothing special. I will do more tests and report back.

uhwsin, 2016-02-27:
You can try a high speed disk . I think hackrf_transfer will stop while the disk speed is not enough .

2i3r, 2016-12-23:
Same happens to me, compiled and installed host from latest master which I cloned from the repo (not sure about the version), on a raspberry pi 2 model B running  4.1.15-3-ARCH.

I tested `dd if=/dev/zero of=./file bs=1M count=100` and it writes with 132 MB/s.
but hackrf_transfer both sample rate 8M and 20M fails after random time (mostly ~100MB ~200MB). 

Just checked the hackrf_transfer code, not sure if it may related to syncing disk. maybe adding `fflush()` after writing good amount of RX may help, I'm gonna test that.

Do I missing something?! any idea how to implement it efficiently?

schneider42, 2016-12-23:
@2i3r: Can you try this branch: https://github.com/rad1o/hackrf/tree/streaming (streaming branch)?

You can supply a buffer size with the `-S` option (ideally a power of two, e.g. 4194304 for 4 MiB). I've seen cases where this helped with sustaining higher rates.

The output will look a bit different, it does not print anything to `stdout` (i.e. no `40.0 MiB / 1.000 sec = 40.0 MiB/second`), expect if it can not write samples to the output anymore.

We'd appreciate if you could give this a try :)

2i3r, 2016-12-23:
@schneider42 Thanks for fast reply :)
Same happens, `hackrf_transfer -r- -f ${FREQ} -s 8000000` sample rate, stops after ~100MB write.

schneider42, 2016-12-23:
Can you try the `-w` option do write directly to a file. Does the tool then abort with an error message?

uhwsin, 2016-12-23:
Monitor your ram  memory usage when you trying hackrf_transfer, maybe you can find something useful.---hackrf_transfer will use ram while the disk speed is not enough.

2i3r, 2016-12-23:
Ok, I just added `fflush()` after each 1MB data written, to the original hackrf_transfer version. tested both (streaming & original with periodical `fflush()`) with `-w` option, The problem is not resolved, but it seems `fflush()` version goes a bit further, at least in couples of tests I've made. 

There is no error message, it seems `hackrf_transfer` aborts streaming when it sees first 0 MiB/second written. This is the output:

> ...
> ...
> Stop with Ctrl-C
> 16.0 MiB / 1.000 sec = 16.0 MiB/second
> 16.0 MiB / 1.000 sec = 16.0 MiB/second
> 14.9 MiB / 1.000 sec = 14.9 MiB/second
> 14.7 MiB / 1.000 sec = 14.7 MiB/second
> 14.7 MiB / 1.000 sec = 14.7 MiB/second
> 14.7 MiB / 1.000 sec = 14.7 MiB/second
>  9.2 MiB / 1.000 sec =  9.2 MiB/second
> 16.3 MiB / 1.000 sec = 16.2 MiB/second
> 16.0 MiB / 1.000 sec = 16.0 MiB/second
> 16.0 MiB / 1.000 sec = 16.0 MiB/second
> 13.9 MiB / 1.000 sec = 13.9 MiB/second
>  0.0 MiB / 1.000 sec =  0.0 MiB/second
> 
> Couldn't transfer any bytes for one second.
> 
> Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
> Total time: 12.00352 s
> hackrf_close() done
> hackrf_exit() done
> fclose(fd) done
> exit

@uhwsin This is really interesting, I monitored the ram using `free -mh`, and it seems files is actually writes to ram not the disk and even not the swap memory, `free` field of `free -mh` decreases as the file size increase, and it gains its free size back when I remove generated iq file. but not when `sync` command used. 
Another insteresting things which happens is one of CPU cores runs with full load when streaming to file but not when redirecting to `/dev/null`. CPU core changes when local minima happens in write speed, (in above log when `9.2 MiB/second` happened). 
So, I tested `dd` again with mush higher size, `dd if=/dev/zero of=~/2TB.dat bs=1M count=2000`. `dd` also writes to ram, until it decreases to 4M-5M, then continue with much lower speed, and this time finish process with 6.6 MB/s average speed. 👍 , then it seems to be something with my rpi2. 
Any idea why the write speed so low?

uhwsin, 2016-12-24:
rpi2 use TF card as it's hard disk, for most TF card, the write speed is slower than read speed. So it is difficult improve the write speed on rpi2.  On the other hand , you can use rpi2+hackrf as a transmitter not a  data acquisition. The read speed is fast enough.  
BTW, use a lower sampler rate(such as 5Msps) may be work fine if you want it to be a  data acquisition yet.

uhwsin, 2016-12-24:
another way, you can try to mount a usb 3.0 USB disk as the hard disk, although rpi2 does not have the usb 3.0 port.

2i3r, 2016-12-24:
I managed to log the 8Msps RX continuously, using a middleware which compress and write data to file in real-time. 
`hackrf_transfer -r- -f ${FREQ} -s 8000000 | zstdmt -t4 > ~/file.iq.zst`
Everything goes fine and it logs unstop even when I restart it several times in a row, but it's not the case all times, somehow in certain times it fails just as like I've mentioned above. For now I don't know what causes this error as the actual data rate using compression should be lower than 11MB/s dsync data write speed on my usb2.0.
Interesting part is were I log 20Msps RX using real-time compression, it need more CPU resource and cause the `hackrf_transfer` to log with lower than normal 40MiB/s speed, ~35MiB/s when level of compression is 1 (fastest), and ~30MiB/s when compression runs at level 3. 
My point is when there is not enough disk write speed we meet 0MiB/s which cause `hackrf-tranfer` to stop, but with not enough CPU power we just see lower data rate, and probably some data periodically drops at board buffer.
@uhwsin thanks wise advice, It should help.

Steve135, 2017-02-07:
Hello, (here is the solution for Windows users, maybe you can do the same in Linux?)
_____________________________________________________________________________________________________
For a previous speaker, who has nothing to do with the length of the cable, possibly with the quality, yes. 
I have a 15 year old, 3m long USB extension cable in use at HackRF, and that works perfectly.
_____________________________________________________________________________________________________
It annoyed me for weeks, that I always only 1x the recording could start.
On Windows8.1 with terminal "**cmder_mini**", after the first recording, then the second started, always came the message:
"**Could not transfer any bytes for one second**"
I had to press the hackRF once the "reset button", then it worked again .... **:-(**
 **I thought: THIS CAN NOT BE THE SOLUTION !!!! The solution, the caching !!!!** **:-)**
 
**1.)** in the system control, "**SYSTEM**", then "**extended system settings**"
**2.)** Click "**Settings**" Button at: "**Visual Effects, Processor Scheduling, Memory Usage and Virtual Memory**"
**3.)** to "**expanded**", then "**virtual memory**", then click on "**change**", "**manual configuration**"

(I have 2x8 = 16 Gb memory in the laptop, and a 500 Gb SSD) So I entered for drive **C:** instead of the **2000 Mb "automatic virtual memory usage", 6000 Mb for C:**. Compter reboot, that's it.
_____________________________________________________________________________________________________
Now it works with me without this error message, no matter, no problem, even with a **sampling rate of 20 MiB/second**, (40 Mb per second), without Probs! 

I have run for over 3 minutes, so about 8.4 gigabytes, the whole thing is repeated, repeated and repeated, many many times, and not even once does this clipped message. :-)
 
By the way, I have also had the RTL-SDR Stick active in GQRX parallel run at the same time.
(For Windows users: first install "**Microsoft Visual C++ 2015 Redistributable**", then "**Python 2.7 for Windows**", and then the "**POTHOS** package". 
This package includes GQRX, the CUBIC-SDR and GNURADIO!

I hope I was able to help you just a little bit...
Thank you for your attention and for reading.
greeting

ZeroChaos-, 2017-07-27:
if it helps, I have this issue with hackrf_sweep too:

```
gato ~ # hackrf_sweep -a 0 -f 400:480 -l 32 -g 32 -w 10000
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 400 MHz to 480 MHz
call hackrf_set_amp_enable(0)
Stop with Ctrl-C
 0.0 MiB / 1.000 sec =  0.0 MiB/second

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: streaming terminated (-1004)
Total time: 1.00015 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
```

ZeroChaos-, 2017-07-27:
even with a saner -w value
```
gato ~ # hackrf_sweep -a 0 -f 400:480 -l 32 -g 32 -w 1000000
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 400 MHz to 480 MHz
call hackrf_set_amp_enable(0)
Stop with Ctrl-C
2017-07-27, 16:38:48, 400000000, 405000000, 1000000.00, 20, -44.57, -46.14, -42.65, -49.56, -57.20
2017-07-27, 16:38:48, 410000000, 415000000, 1000000.00, 20, -50.80, -51.51, -54.88, -47.39, -46.56
2017-07-27, 16:38:48, 405000000, 410000000, 1000000.00, 20, -46.30, -71.12, -54.01, -48.92, -45.21
2017-07-27, 16:38:48, 415000000, 420000000, 1000000.00, 20, -45.33, -56.83, -42.70, -44.61, -51.96
2017-07-27, 16:38:48, 420000000, 425000000, 1000000.00, 20, -43.80, -47.06, -42.89, -43.70, -61.86
2017-07-27, 16:38:48, 430000000, 435000000, 1000000.00, 20, -52.50, -53.64, -43.01, -44.25, -45.62
2017-07-27, 16:38:48, 425000000, 430000000, 1000000.00, 20, -55.10, -47.28, -46.28, -47.30, -54.48
2017-07-27, 16:38:48, 435000000, 440000000, 1000000.00, 20, -46.72, -55.27, -48.72, -38.76, -32.74
2017-07-27, 16:38:48, 440000000, 445000000, 1000000.00, 20, -48.14, -47.86, -47.54, -47.56, -47.85
2017-07-27, 16:38:48, 450000000, 455000000, 1000000.00, 20, -28.21, -12.21, -4.96, -7.90, -25.98
2017-07-27, 16:38:48, 445000000, 450000000, 1000000.00, 20, -51.23, -48.55, -49.04, -49.59, -45.13
2017-07-27, 16:38:48, 455000000, 460000000, 1000000.00, 20, -39.37, -40.51, -49.04, -49.11, -67.40
2017-07-27, 16:38:48, 460000000, 465000000, 1000000.00, 20, -38.65, -47.22, -36.05, -40.01, -45.22
2017-07-27, 16:38:48, 470000000, 475000000, 1000000.00, 20, -49.25, -38.62, -32.21, -34.61, -42.69
2017-07-27, 16:38:48, 465000000, 470000000, 1000000.00, 20, -52.22, -46.45, -57.39, -44.10, -39.20
2017-07-27, 16:38:48, 475000000, 480000000, 1000000.00, 20, -40.82, -41.39, -49.76, -43.96, -43.53
2017-07-27, 16:38:48, 400000000, 405000000, 1000000.00, 20, -43.51, -63.42, -50.71, -54.93, -47.90
2017-07-27, 16:38:48, 410000000, 415000000, 1000000.00, 20, -51.79, -46.91, -44.97, -46.43, -49.63
2017-07-27, 16:38:48, 405000000, 410000000, 1000000.00, 20, -47.46, -40.59, -41.27, -45.85, -47.47
2017-07-27, 16:38:48, 415000000, 420000000, 1000000.00, 20, -48.29, -53.70, -48.89, -42.40, -41.37
2017-07-27, 16:38:48, 420000000, 425000000, 1000000.00, 20, -55.31, -46.57, -48.41, -43.58, -42.31
2017-07-27, 16:38:48, 430000000, 435000000, 1000000.00, 20, -44.17, -46.58, -46.99, -55.06, -44.80
2017-07-27, 16:38:48, 425000000, 430000000, 1000000.00, 20, -52.30, -52.92, -55.42, -46.37, -51.67
2017-07-27, 16:38:48, 435000000, 440000000, 1000000.00, 20, -49.42, -45.29, -45.39, -47.86, -38.01
2017-07-27, 16:38:48, 440000000, 445000000, 1000000.00, 20, -44.48, -55.95, -45.12, -54.88, -43.84
2017-07-27, 16:38:48, 450000000, 455000000, 1000000.00, 20, -26.62, -18.96, -6.38, -7.81, -23.69
2017-07-27, 16:38:48, 445000000, 450000000, 1000000.00, 20, -45.90, -44.66, -47.90, -53.13, -39.02
2017-07-27, 16:38:48, 455000000, 460000000, 1000000.00, 20, -35.02, -46.12, -50.66, -44.10, -45.65
2017-07-27, 16:38:48, 460000000, 465000000, 1000000.00, 20, -38.21, -41.26, -45.52, -46.76, -45.25
2017-07-27, 16:38:48, 470000000, 475000000, 1000000.00, 20, -41.93, -36.07, -30.08, -33.90, -47.41
2017-07-27, 16:38:48, 465000000, 470000000, 1000000.00, 20, -41.99, -49.17, -46.86, -42.02, -42.33
2017-07-27, 16:38:48, 475000000, 480000000, 1000000.00, 20, -50.19, -44.48, -45.72, -40.39, -44.28
2017-07-27, 16:38:48, 400000000, 405000000, 1000000.00, 20, -61.20, -46.14, -42.76, -41.26, -38.48
2017-07-27, 16:38:48, 410000000, 415000000, 1000000.00, 20, -49.24, -45.08, -42.03, -46.74, -44.81
2017-07-27, 16:38:48, 405000000, 410000000, 1000000.00, 20, -58.02, -54.51, -47.23, -44.10, -44.03
2017-07-27, 16:38:48, 415000000, 420000000, 1000000.00, 20, -50.39, -40.16, -42.99, -43.67, -43.22
2017-07-27, 16:38:48, 420000000, 425000000, 1000000.00, 20, -40.53, -40.13, -37.65, -37.99, -43.51
2017-07-27, 16:38:48, 430000000, 435000000, 1000000.00, 20, -46.79, -46.10, -41.38, -39.56, -34.15
2017-07-27, 16:38:48, 425000000, 430000000, 1000000.00, 20, -44.41, -50.67, -37.85, -39.39, -47.51
2017-07-27, 16:38:48, 435000000, 440000000, 1000000.00, 20, -45.46, -36.88, -35.66, -44.57, -41.64
2017-07-27, 16:38:48, 440000000, 445000000, 1000000.00, 20, -42.83, -45.09, -45.37, -43.39, -43.83
2017-07-27, 16:38:48, 450000000, 455000000, 1000000.00, 20, -26.90, -9.76, -4.06, -7.94, -29.25
2017-07-27, 16:38:48, 445000000, 450000000, 1000000.00, 20, -42.48, -55.46, -48.61, -51.36, -48.63
2017-07-27, 16:38:48, 455000000, 460000000, 1000000.00, 20, -39.90, -53.67, -45.39, -50.98, -43.44
2017-07-27, 16:38:48, 460000000, 465000000, 1000000.00, 20, -43.57, -47.79, -50.31, -43.29, -45.54
2017-07-27, 16:38:48, 470000000, 475000000, 1000000.00, 20, -43.38, -38.43, -31.54, -34.22, -56.05
2017-07-27, 16:38:48, 465000000, 470000000, 1000000.00, 20, -36.98, -49.49, -52.91, -45.23, -42.96
2017-07-27, 16:38:48, 475000000, 480000000, 1000000.00, 20, -41.80, -46.58, -63.67, -57.78, -42.59
2017-07-27, 16:38:48, 400000000, 405000000, 1000000.00, 20, -63.07, -52.18, -57.51, -47.21, -47.05
2017-07-27, 16:38:48, 410000000, 415000000, 1000000.00, 20, -45.45, -45.92, -43.52, -47.92, -47.04
2017-07-27, 16:38:48, 405000000, 410000000, 1000000.00, 20, -47.85, -41.97, -41.29, -49.10, -49.50
2017-07-27, 16:38:48, 415000000, 420000000, 1000000.00, 20, -47.19, -47.18, -63.87, -46.99, -46.75
[... cut ...]
2017-07-27, 16:38:48, 450000000, 455000000, 1000000.00, 20, -28.73, -18.76, -6.50, -8.10, -25.23
2017-07-27, 16:38:48, 445000000, 450000000, 1000000.00, 20, -42.90, -43.42, -47.92, -50.12, -40.42
2017-07-27, 16:38:48, 455000000, 460000000, 1000000.00, 20, -50.17, -41.64, -43.00, -45.10, -43.68
 0.5 MiB / 1.000 sec =  0.5 MiB/second

Exiting... hackrf_is_streaming() result: streaming terminated (-1004)
Total time: 1.00008 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
```

dominicgs, 2017-09-13:
@ZeroChaos- Could you re-test this with the latest firmware?  I had someone report this issue to me this week and it was fixed by updating the firmware to the latest git firmware.

I'm not suggesting that I've intentionally fixed this issue, as I haven't been able to reproduce it yet.

elal007, 2017-10-17:
I have the same problem AFTER I upgrade the firmware with the latest . Now, I am looking how to solve this issue. My upgraded actually succes, i have -H options in the hackrf_transfer.
elal@ubuntu:~$ hackrf_transfer -d 0000000000000000925866e62d666ee3 -t transferfile.txt -f 2412000000 -l 24 -g 20 -s 8000000000 -b 1750000 -R 1
call hackrf_set_sample_rate(3705032704 Hz/3705.033 MHz)
call hackrf_baseband_filter_bandwidth_set(1750000 Hz/1.750 MHz)
call hackrf_set_hw_sync_mode(0)
call hackrf_set_freq(2412000000 Hz/2412.000 MHz)
Stop with Ctrl-C
 0.0 MiB / 1.000 sec =  0.0 MiB/second

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 1.00011 s
hackrf_stop_tx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit

dominicgs, 2017-10-18:
@elal007 Could you show the output of `hackrf_info` and try using `hackrf_transfer -r/dev/null`?

elal007, 2017-10-18:
Hi Dominicgs,
I solve my problem by try upgrade firmware again using 
elal@ubuntu:~/Downloads/hackrf-2017.02.1/firmware-bin$ dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu

Then, I install the firmware again

elal@ubuntu:~/Downloads/sdr-utils-master$ hackrf_info
hackrf_info version: git-2174aa0
libhackrf version: git-2174aa0 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000925866e62d666ee3
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00694f5d
elal@ubuntu:~/Downloads/sdr-utils-master$ hackrf_transfer -r/dev/null
call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
call hackrf_set_hw_sync_mode(0)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
19.9 MiB / 1.000 sec = 19.9 MiB/second
19.9 MiB / 1.000 sec = 19.9 MiB/second
19.9 MiB / 1.000 sec = 19.9 MiB/second
20.2 MiB / 1.000 sec = 20.2 MiB/second
19.9 MiB / 1.000 sec = 19.9 MiB/second
^CCaught signal 2
 8.1 MiB / 0.403 sec = 20.2 MiB/second

Exiting...
Total time: 5.40335 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit

Using hackrf_sweep
2017-10-18, 02:44:03.860698, 2920000000, 292500^CCaught signal 2
5 total sweeps completed, 1.23 sweeps/second

Exiting...
Total sweeps: 5 in 4.05981 seconds (1.23 sweeps/second)
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
0000, 1000000.00, 20, -104.06, -100.08, -96.01, -91.41, -85.57
2017-10-18, 02:44:03.860698, 2930000000, 2935000000, 1000000.00, 20, -85.57, -91.41, -96.01, -100.08, -104.06
2017-10-18, 02:44:03.860698, 2925000000, 2930000000, 1000000.00, 20, -65.04, -73.15, -67.37, -61.54, -72.59
2017-10-18, 02:44:03.860698, 2935000000, 2940000000, 1000000.00, 20, -71.69, -76.74, -68.29, -64.73, -66.45
2017-10-18, 02:44:03.860698, 2940000000, 2945000000, 1000000.00, 20, -63.68, -62.09, -63.91, -71.37, -79.66
2017-10-18, 02:44:03.860698, 2950000000, 2955000000, 1000000.00, 20, -**63.27, -62.28, -63.73, -72.02, -78.34**
fclose(fd) done
exit
Btw, what is value after 20Msps??. I guess it is the RSSI from transmitter??. Am i correct?

Thanks you.

dominicgs, 2017-10-18:
The values at the end of each line are the power level for each bin of the FFT that we perform on the data.  It's similar to RSSI but calculated in a different way.

Brainticket, 2017-11-05:
Hello ! 
Same problem here ! New hackrf one - delivered two days ago.
Yesterday I installed new firmware / libs and tools from git...

> alex@ubuntu:~/dev/hackrf/firmware/hackrf_usb/build$ hackrf_info 
> hackrf_info version: git-f0473b7
> libhackrf version: git-f0473b7 (0.5)
> Found HackRF
> Index: 0
> Serial number: 0000000000000000a06063c8234a325f
> Board ID Number: 2 (HackRF One)
> Firmware Version: git-f0473b7 (API:1.03)
> Part ID Number: 0xa000cb3c 0x006e4f5d

When  I start hackrf_transfer this will only work for one time : 

> alex@ubuntu:~/dev/hackrf/firmware/hackrf_usb/build$ hackrf_transfer -r/dev/null
> call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
> call hackrf_set_hw_sync_mode(0)
> call hackrf_set_freq(900000000 Hz/900.000 MHz)
> Stop with Ctrl-C
> 19.9 MiB / 1.001 sec = 19.9 MiB/second
> 19.9 MiB / 1.000 sec = 19.9 MiB/second
> 20.2 MiB / 1.001 sec = 20.2 MiB/second
> 19.9 MiB / 1.000 sec = 19.9 MiB/second
> 19.9 MiB / 1.001 sec = 19.9 MiB/second
> 20.2 MiB / 1.001 sec = 20.2 MiB/second
> 19.9 MiB / 1.000 sec = 19.9 MiB/second
> 19.9 MiB / 1.001 sec = 19.9 MiB/second
> 20.2 MiB / 1.001 sec = 20.2 MiB/second
> ^CCaught signal 2
>  5.5 MiB / 0.283 sec = 19.4 MiB/second
> 
> Exiting...
> Total time: 9.28893 s
> hackrf_stop_rx() done
> hackrf_close() done
> hackrf_exit() done
> fclose(fd) done
> exit

When I try to start again,  I get the following error:

> alex@ubuntu:~/dev/hackrf/firmware/hackrf_usb/build$ hackrf_transfer -r/dev/null
> call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
> call hackrf_set_hw_sync_mode(0)
> call hackrf_set_freq(900000000 Hz/900.000 MHz)
> Stop with Ctrl-C
>  0.0 MiB / 1.001 sec =  0.0 MiB/second
> 
> Couldn't transfer any bytes for one second.
> 
> Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
> Total time: 1.00109 s
> hackrf_stop_rx() done
> hackrf_close() done
> hackrf_exit() done
> fclose(fd) done
> exit

**Only a reset (button / power off/on) can bring back the hackrf to talk with me :(**

I tried different sample rates without success...
I this possible a hardware defect ?

Thanks !

elal007, 2017-11-05:
HI @Brainticket ,
Have you tried to reinstall a new firmware again using DFU functions:
elal@ubuntu:~/Downloads/hackrf-2017.02.1/firmware-bin$ dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu

Brainticket, 2017-11-05:
Hello @elal007 ,
Yes - I tried this way too - without success ...

Greets

dominicgs, 2017-11-07:
@Brainticket Can I confirm (to make sure I have this correct) that:
 * you are running the latest firmware and host code from git?
 * the first time you perform an operation it succeeds but the subsequent times give this issue?

Brainticket, 2017-11-08:
@dominicgs 

- you are running the latest firmware and host code from git?
**yes**

- the first time you perform an operation it succeeds but the subsequent times give this issue?
**yes**

But ... I tried everything in vmware workstation !
Yesterday I installed a clean Ubuntu notebook and build firmware & tools again - now ereything seems to be ok ... 
After that I build the tools for Win10 with cygwin - seems to be ok too !

But what is the problem with vmware ( Workstation Pro / 14.0.0 build-6661328) ? 
Why does transfer only work for the first time ?

dominicgs, 2017-11-09:
We've often had problems with VMs having flaky USB support when it comes to high throughput, although I've heard very few complaints when it comes to vmware.

It may be that we're doing something that vmware doesn't like.  It's only falling over on the bulk transfers, not the control transfers, so that's probably a clue.

ZeroChaos-, 2018-08-21:
I am no longer experiencing this issue.  I suspect at least for my case that overzealous powertop settings and insufficient hardware were the cause.  I am still having #362 though

Commander-Crash, 2020-01-16:
This issue with the "Couldn't transfer any bytes for one second" I suspect it's the USB connection. I get this when I use it with a raspberry pi 3 B. When I have it connected to my desktop I have no issues. I removed USB sound card from my pi3-B And had no future issues. I hope this helps.

closed 2020-02-12 by miek