greatscottgadgets/hackrf #238: hack_transfer on windows dropping samples

mbavaro, 2016-02-04:
I compiled hack_transfer on mingw32 to my best knowledge.. however it drops samples from time to time (in 10 minutes it can happen a few times) when transmitting at 10Msps.
curiously this does not happen on Linux.
With Zadig I replaced the WinUSB driver to libusb (1.0.20 from libusb.info) but then I get a USB error setting the bandwidth. Any clues anyone?

Mic

mbavaro, 2016-02-19:
OK, thinking about it I found the cause. RX is very stable even at high sampling rates, whilst TX is not even at low sampling rates. The disk will be very capable of buffering writes, but not equally capable of reliably fetching the data when it is needed. The solution I found (not thread safe, but easy to make so) is to use a large circular buffer.

circbuff.h
```
#ifndef _CIRCUBUFF_H_
#define _CIRCUBUFF_H_

int InitCircBuffer(const char _cFileName[], int _iSize);
int RefillCircBuffer(void);
int ReadFromBuffer(void *_pMem, int _iReadSize);
int CleanCircBuffer(void);

#endif /* _CIRCUBUFF_H_ */
```

circbuff.c
```
#include <stdlib.h>
#include <stdint.h>
#include <string.h>
#include <stdio.h>

static FILE *fid;
static int iBuffSize;
static int8_t *pBuff;
static int iBuffHead, iBuffTail;

/** InitCircBuffer */
int InitCircBuffer(const char _cFileName[], int _iSize) {
  
  fid = fopen(_cFileName, "rb");
  if (NULL == fid) {
    fprintf(stderr, "error opening input file %s\n", _cFileName);
    goto CLEANUP;
  }
  pBuff = malloc(_iSize);
  if (NULL == pBuff) {
    fprintf(stderr, "error allocating memory buffer of size %.1f kBytes\n", _iSize/1024.0);
    goto CLEANUP;
  }
  iBuffSize = _iSize;
  iBuffHead = 0;
  iBuffTail = fread(pBuff, 1, 3*_iSize/4, fid);      
  fprintf(stderr,"size %d head %d tail %d\n", iBuffSize, iBuffHead, iBuffTail);
  
  return 0;  
  
  CLEANUP:
  if (NULL != fid) {
    fclose(fid);
    fid = NULL;
  }  
  if (NULL != pBuff) {
    free(pBuff);
    pBuff = NULL;
  }   
  return -1;
}

/** RefillCircBuffer */
int RefillCircBuffer(void) {
  int iToRead, iLeftOver, iRet = 0;
  
  if (iBuffHead<iBuffTail) {
    iToRead = iBuffSize -(iBuffTail-iBuffHead);
  } else {
    iToRead = iBuffHead-iBuffTail;
  }
  fprintf(stderr,"refilling %d kBytes\n", iToRead/1024);
  if (iToRead > 0) {
    if ((iBuffTail+iToRead) > iBuffSize) {
      iRet  = fread(pBuff+iBuffTail, 1, iBuffSize-iBuffTail, fid);
      iLeftOver = (iBuffTail+iToRead) -iBuffSize;
      iRet += fread(pBuff          , 1, iLeftOver,           fid);
    } else {
      iRet = fread(pBuff+iBuffTail, 1, iToRead, fid);
    }
    iBuffTail = (iBuffTail+iRet) % iBuffSize;
  }
  return iRet;
}

/** ReadFromBuffer */
int ReadFromBuffer(void *_pMem, int _iReadSize) {
  int iAvail, iLeftOver;
  int8_t *pCastMem = (int8_t*) _pMem;
  
  if (iBuffHead<=iBuffTail) {
    iAvail =             iBuffTail-iBuffHead;
  } else {
    iAvail = iBuffSize -(iBuffHead-iBuffTail);
  }
  
  if (_iReadSize > iAvail) {
    _iReadSize = iAvail;
  }
  
  if ((iBuffHead+_iReadSize) > iBuffSize) {
    memcpy(pCastMem,                       pBuff+iBuffHead, iBuffSize-iBuffHead);
    iLeftOver = (iBuffHead+_iReadSize) -iBuffSize;
    memcpy(pCastMem+(iBuffSize-iBuffHead), pBuff,            iLeftOver);    
  } else {
    memcpy(pCastMem,                       pBuff+iBuffHead, _iReadSize);    
  }
  iBuffHead = (iBuffHead+_iReadSize) % iBuffSize;
  return _iReadSize;
}

/** CleanCircBuffer */
int CleanCircBuffer(void) {
  if (NULL != fid) {
    fclose(fid);
    fid = NULL;
  }  
  if (NULL != pBuff) {
    free(pBuff);
    pBuff = NULL;
  }   
  iBuffHead = 0; 
  iBuffTail = 0;
  iBuffSize = 0;
  return 0;
}

```

and modify hack_transfer in the following way:

```
if( transceiver_mode == TRANSCEIVER_MODE_RX )
		{
			fd = fopen(path, "wb");
		} else {
			fd = fopen(path, "rb");
			fclose(fd);
			InitCircBuffer(path, 256*1024*1024);
		}
```

```
	while( (hackrf_is_streaming(device) == HACKRF_TRUE) &&
			(do_exit == false) ) 
	{
		uint32_t byte_count_now;
		struct timeval time_now;
		float time_difference, rate;
		sleep(1);

		if( transceiver_mode == TRANSCEIVER_MODE_TX ) {
			RefillCircBuffer();
		}

		gettimeofday(&time_now, NULL);
```

```
//bytes_read = fread(transfer->buffer, 1, bytes_to_read, fd);
		bytes_read = ReadFromBuffer(transfer->buffer, bytes_to_read);
		if ((bytes_read != bytes_to_read)
				|| (limit_num_samples && (bytes_to_xfer == 0))) {
			if (repeat) {
				printf("Input file end reached. Rewind to beginning.\n");
				rewind(fd);
				// fread(transfer->buffer + bytes_read, 1, bytes_to_read - bytes_read, fd);
				ReadFromBuffer(transfer->buffer + bytes_read, bytes_to_read - bytes_read);
				return 0;
			} else {
				return -1; // not loopback mode, EOF
			}

		} else {
			return 0;
		}
```

```
		if( transmit ) 
		{
			result = hackrf_stop_tx(device);
			if( result != HACKRF_SUCCESS ) {
				printf("hackrf_stop_tx() failed: %s (%d)\n", hackrf_error_name(result), result);
			}else {
				printf("hackrf_stop_tx() done\n");
			}
			CleanCircBuffer();
			fd = NULL;
		}
```

Seems like I am now able to stream with no packet losses at 12.5 Msps

dominicgs, 2016-02-19:
Thanks for hunting this down, I'm sorry we weren't able to help more with it.

I think your solution is good, but I'm not sure if we want to modify hackrf_transfer in that way or if there's another way that this can be handled.  For example, I'd prefer to use Linux's inotify(), but it's not available on other platforms.

mbavaro, 2016-03-08:
Oops, I guess

`if (iBuffHead<=iBuffTail) {
`
in 

`int ReadFromBuffer(...)
`
should be

`if (iBuffHead<iBuffTail) {
`
instead.

mossmann, 2017-01-27:
I'm wondering if the -S streaming option (merged in #239) would fix this if adapted from RX to TX.

mbavaro, 2017-01-27:
Provided the circular buffer is filled prior to start transmission, sure. Actually, that is simply what the code I wrote did ;)

closed 2017-11-17 by dominicgs