greatscottgadgets/hackrf #240: CPLD Flash Failure/No RX

tlmain, 2016-03-07:
My hackrf One has recently stopped receiving any signals at.

- hackrf_info returns valid information with no error
- Updated firmware using version hackrf-2015.07.2 with no problems.
- Attempted to update CPLD and received: _hackrf_cpld_write() failed: HACKRF_ERROR_LIBUSB (-1000)_
- Using grqx (debian) or SDR# (win): A quick "click" is received then complete silence, adjusting gains makes no difference in reception.
- No relevant dmesg entries
- Using powered USB with ample power (3+amp)

Any ideas or suggestions?

dominicgs, 2016-03-10:
Is the 2015.07.2 firmware written to flash on the device, or loaded using DFU mode?  Have you attempted to re-flash it, in case it was corrupted during the write process?

Do other functions work, e.g. turning on the RX amp?
Setting the frequency?
Are you able to test transmit?

When you say that there are no relevant dmesg entries, do you mean that it appears as expected?  If it's enumerating, it should be producing some dmesg output.

Are you using a laptop with an up to date Linux distribution?  Even when powered via an external supply, recent distributions have become more strict about power management and HackRF does not support USB suspend.  The work around is to add it to the power management blacklist.

tlmain, 2016-03-15:
It looks like it was a dead CPLD. Replaced chip and flashed successfully. All is working now.

Just for information:
- 2015.07.2 firmware flashed multiple times successfully in normal and DFU mode
- RX Amp appeared to enable OK
- Frequency set appeared to work
- Transmitting reported it was working but no TX LED, nothing received on other radio
- dmesg reported that the HackRF was detected and configured correctly, no errors/warnings
- Was using latest (8.3 I believe) Debian x64
- Power supplied via USB hub connected to external power supply

closed 2016-03-15 by tlmain

ampledata, 2016-11-16:
FWIW I just ran into the same problem as @tlmain when attempting to flash a Jawbreaker.

```
root@host:/home/gba/src/hackrf-2015.07.2# lsusb
Bus 001 Device 004: ID 1d50:604b OpenMoko, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@host:/home/gba/src/hackrf-2015.07.2# hackrf_info 
Found HackRF board.
Board ID Number: 1 (Jawbreaker)
Firmware Version: 2015.07.2
Part ID Number: 0xbc5f4348 0xbc5f4348
Serial Number: 0x00000000 0x00000000 0x066062c8 0x350b6687
root@host:/home/gba/src/hackrf-2015.07.2# hackrf_spiflash -w firmware-bin/hackrf_jawbreaker_usb_rom_to_ram.bin
File size 19816 bytes.
Erasing SPI flash.
Writing 256 bytes at 0x000000.
Writing 256 bytes at 0x000100.
Writing 256 bytes at 0x000200.
Writing 256 bytes at 0x000300.
Writing 256 bytes at 0x000400.
Writing 256 bytes at 0x000500.
Writing 256 bytes at 0x000600.
Writing 256 bytes at 0x000700.
...
Writing 256 bytes at 0x004400.
Writing 256 bytes at 0x004500.
Writing 256 bytes at 0x004600.
Writing 256 bytes at 0x004700.
Writing 256 bytes at 0x004800.
Writing 256 bytes at 0x004900.
Writing 256 bytes at 0x004a00.
Writing 256 bytes at 0x004b00.
Writing 256 bytes at 0x004c00.
Writing 104 bytes at 0x004d00.
root@host:/home/gba/src/hackrf-2015.07.2# echo $?
0
```

At this point I unplug and re-plug the Jawbreaker:
```
root@host:/home/gba/src/hackrf-2015.07.2# lsusb
Bus 001 Device 006: ID 1d50:604b OpenMoko, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@host:/home/gba/src/hackrf-2015.07.2# hackrf_info 
Found HackRF board.
Board ID Number: 1 (Jawbreaker)
Firmware Version: 2015.07.2
Part ID Number: 0xbc5f4348 0xbc5f4348
Serial Number: 0x00000000 0x00000000 0x066062c8 0x350b6687
root@host:/home/gba/src/hackrf-2015.07.2# hackrf_cpldjtag -x firmware/cpld/sgpio_if/default.xsvf
File size 37629 bytes.
LED1/2/3 blinking means CPLD program success.
LED3/RED steady means error.
Wait message 'Write finished' or in case of LED3/RED steady, Power OFF/Disconnect the Jawbreaker.
hackrf_cpld_write() failed: HACKRF_ERROR_LIBUSB (-1000)
```

ampledata, 2016-11-16:
Found the problem. I was flashing with an old version of the hackrf tools (2013). Re-tried with latest tools (2015.07.2) and all is good.