greatscottgadgets/hackrf #242: Hi guys

darmoy, 2016-03-23:
I was so thrilled when my gf bought me a HackRF as a birthday present. My bday isn't till the 25 but i couldn't wait no more. She is sleeping (Eastern Europe here)
So i sneaked out of bed and unbox and plug my hackrf

I was mega excited tonight. After installing all the lib and the latest firmware for my hackrf, i shoot:

```
livebullshit@Alien-LBS:~$ hackrf_info 
Found HackRF board 0:
USB descriptor string: 0000000000000000406464c8236b804b
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2
Part ID Number: 0xa000cb3c 0x00614349
Serial Number: 0x00000000 0x00000000 0x406464c8 0x236b804b
livebullshit@Alien-LBS:~$ 
```
So far so good. First thing i try is to check the hackrf with gqrx and tune into random FM.

**And here come the big issue **

Picture worth 10 millions words:

[http://5.135.185.152/sdr/hackrf_broken_rxtx.jpg](url)

The result is pure  .... FLATLINE. i'm not a SDR guru but neither a noob. 
To be sure (maybe a gqrx bug. so i go for grc and create a simple fm rx on a freq i know there is something). Same result. 
almost having an heart attack now. Nothing being received by my hackrf.
But i'm persistent, i switch to windows gf laptop, zadig it for the driver and shoot sdrsharp....still flatline.

Btw, i tested TX and it's the same... nothing :(

Last test, i plug a RTLSDR dongle to check and miracle:

[http://5.135.185.152/sdr/rtlsdr_rx.jpg](url)

So i really don't understand what's happening.

Anyone having an idea.

jboone, 2016-03-23:
For what it's worth, the image you sent is not a flat line. It contains noise, and has the ubiquitous DC spike, and apparently some spur at 100MHz. So *some* of the HackRF is working fine. It's possible the antenna you have is shorted, and so you're only seeing internally-generated noise. I would first try a different antenna -- even a bent paper clip stuck into the center hole of the SMA would be sufficient to see FM broadcast stations.

It would also be telling that the noise floor increases or decreases as you change the RF and baseband gains. I suspect they will change.

	- Jared

> On Mar 22, 2016, at 20:58, Guillaume <notifications@github.com> wrote:
> 
> The result is pure .... FLATLINE. i'm not a SDR guru but neither a noob.

darmoy, 2016-03-23:
Using a bent paper clip indeed works and display the FM. I only have my ANT500 so can't test other antenna. Like the rest of u said happened.
What does this mean ? antenna shorted ?

closed 2016-03-23 by darmoy

detjar, 2018-02-08:
Hi

I experience almost the same issue, except that bent paper clip does not help at all. Could you share with me, what was the issue in your case?