greatscottgadgets/hackrf #243: HACKRF_ERROR_LIBUSB (-1000) issue

Zyg0te, 2016-03-27:
Trying to get the HackRF working in Arch Linux. Ive installed gqrx which pulls in the necessary hackrf packages. The problem is that gqrx cant detect the hackrf and hackrf_info returns the following error (with debug enabled)

```
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/010: Broken pipe
Found HackRF board 0:
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/010: No such device
hackrf_open() failed: HACKRF_ERROR_LIBUSB (-1000)
```

If I run hackrf_info again within the next three seconds, I get the following:

```
Found HackRF board 0:
USB descriptor string: 0000000000000000406464c82392384b
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2
Part ID Number: 0xa000cb3c 0x00674757
Serial Number: 0x00000000 0x00000000 0x4(REMOVED)
```

I get the same behavior whether Im running as root or not. 

Any ideas?

dominicgs, 2016-03-29:
This looks like it may be due to power saving being active on your system.

I have blacklisted suto suspend for HackRF locally by adding a file called `/etc/laptop-mode/conf.d/hackrf.conf` with the contents:
```
# HackRF One does not support suspend
AUTOSUSPEND_USBID_BLACKLIST="1d50:604b"
AUTOSUSPEND_USBID_BLACKLIST="1d50:6089"
```

I would expect this to fix your issue, although there are multiple power saving tools which require different methods.

Zyg0te, 2016-03-29:
Ah, awesome. I'm currently using [TLP](https://wiki.archlinux.org/index.php/TLP) so I'll have to figure out how to blacklist auto suspend there, but it's very possible that power saving is the reason I'm having the issue.

Blacklisting the way you do will only disable auto suspend for the hackrf though, right?

dominicgs, 2016-03-29:
Correct. You blacklist by USB ID.

I think I have a sample tlp config file somewhere but it will take about an
hour before I can get to it. It's very similar to the laptpp-mode version.

Zyg0te, 2016-03-29:
Awesome. I wont get to verify the fix until I get home later today anyway. I would greatly appreciate a TLP-config sample ;)

dominicgs, 2016-03-29:
The file is `/etc/default/tlp`, it probably already exists, so you should only need to add the following to it
```
USB_BLACKLIST="1d50:6089 1d50:604b"
```
It's possible that you already have a `USB_BLACKLIST` line, so you will likely want to modify that rather than adding a new one.

dominicgs, 2016-07-20:
@Zyg0te Did this solve the problem for you?

I would like to close this issue if the problem was solved.

petterreinholdtsen, 2016-10-29:
I see the same problem with a Thinkpad X230, and while modifying /etc/default/tlp helped me to get gqrx working with the HackRF One for the first time, I was not able to get it working when I tried the second time.

Here is a demonstration just after pressing the reset button:

```
root@minerva:~# while hackrf_info; do sleep 1; done
Found HackRF board.
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2
Part ID Number: 0xa000cb3c 0x00534f59
Serial Number: 0x00000000 0x00000000 0x909864c8 0x396e46cf
Found HackRF board.
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2
Part ID Number: 0xa000cb3c 0x00534f59
Serial Number: 0x00000000 0x00000000 0x909864c8 0x396e46cf
hackrf_open() failed: HACKRF_ERROR_LIBUSB (-1000)
root@minerva:~# 
```

This happen consistently, and some times I do not manage to get any successful hackrf_info run before it fail.

dominicgs, 2016-10-29:
Do you have `laptop-mode` installed too?

Have you restarted tlp since modifying the config?

petterreinholdtsen, 2016-10-29:
I do not have laptop-mode installed?  I used to have laptop-mode-tools installed, but not any more:
```
root@minerva:~# dpkg -l |grep laptop-mode
rc  laptop-mode-tools  1.66-2 all  Tools for Power Savings based on battery/AC status
root@minerva:~# 
```
I ran 'service tlp stop; service tlp start' to allow the blacklist change to take effect.  Before I did this, I could not get gqrx to work at all.

dominicgs, 2016-10-29:
Could you stop tlp entirely (`service tlp stop`) and see if the problem goes away?  I'd like to determine if this is a USB power saving issue or something else.

petterreinholdtsen, 2016-10-30:
I tried, and it did not make a difference:

```
root@minerva:~# service tlp stop
[ inserting hackrf in usb ]
root@minerva:~# while hackrf_info; do sleep 1; done
Found HackRF board.
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2
Part ID Number: 0xa000cb3c 0x00534f59
Serial Number: 0x00000000 0x00000000 0x909864c8 0x396e46cf
Found HackRF board.
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2
Part ID Number: 0xa000cb3c 0x00534f59
Serial Number: 0x00000000 0x00000000 0x909864c8 0x396e46cf
Found HackRF board.
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2
Part ID Number: 0xa000cb3c 0x00534f59
Serial Number: 0x00000000 0x00000000 0x909864c8 0x396e46cf
hackrf_open() failed: HACKRF_ERROR_LIBUSB (-1000)
root@minerva:~#
```

Should I try using LIBUSB_DEBUG to get more output?  I checked /var/log/syslog, and there are no messages from the kernel then hackrf_info get stuck.

dominicgs, 2016-10-30:
That suggests that tlp isn't the issue.  Running with LIBUSB_DEBUG set would be great.

petterreinholdtsen, 2016-10-30:
[Dominic Spill]
> That suggests that tlp isn't the issue.  Running with LIBUSB_DEBUG set
> would be great.

Here is the last few lines from 'while LIBUSB_DEBUG=7 hackrf_info; do
sleep 1; done':

libusb: debug [linux_get_device_address] bus=4 dev=1
libusb: debug [linux_enumerate_device] busnum 4 devaddr 1 session_id 1025
libusb: debug [linux_enumerate_device] allocating new device for 4/1 (session 1025)
libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
libusb: debug [linux_get_device_address] scan usb1
libusb: debug [linux_get_device_address] bus=1 dev=1
libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
libusb: debug [linux_get_device_address] scan 1-1
libusb: debug [linux_get_device_address] bus=1 dev=2
libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
libusb: debug [linux_get_parent_info] Dev 0xfbd980 (1-1) has parent 0xfbd8c0 (usb1) port 1
libusb: debug [linux_get_device_address] getting address for device: 1-1.2 detached: 0
libusb: debug [linux_get_device_address] scan 1-1.2
libusb: debug [linux_get_device_address] bus=1 dev=75
libusb: debug [linux_enumerate_device] busnum 1 devaddr 75 session_id 331
libusb: debug [linux_enumerate_device] allocating new device for 1/75 (session 331)
libusb: debug [linux_get_parent_info] Dev 0xfbf2f0 (1-1.2) has parent 0xfbd980 (1-1) port 2
libusb: debug [linux_get_device_address] getting address for device: 1-1.3 detached: 0
libusb: debug [linux_get_device_address] scan 1-1.3
libusb: debug [linux_get_device_address] bus=1 dev=3
libusb: debug [linux_enumerate_device] busnum 1 devaddr 3 session_id 259
libusb: debug [linux_enumerate_device] allocating new device for 1/3 (session 259)
libusb: debug [linux_get_parent_info] Dev 0xfbcaa0 (1-1.3) has parent 0xfbd980 (1-1) port 3
libusb: debug [linux_get_device_address] getting address for device: 1-1.6 detached: 0
libusb: debug [linux_get_device_address] scan 1-1.6
libusb: debug [linux_get_device_address] bus=1 dev=5
libusb: debug [linux_enumerate_device] busnum 1 devaddr 5 session_id 261
libusb: debug [linux_enumerate_device] allocating new device for 1/5 (session 261)
libusb: debug [linux_get_parent_info] Dev 0xfbcb90 (1-1.6) has parent 0xfbd980 (1-1) port 6
libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
libusb: debug [linux_get_device_address] scan usb2
libusb: debug [linux_get_device_address] bus=2 dev=1
libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
libusb: debug [linux_get_device_address] scan 2-1
libusb: debug [linux_get_device_address] bus=2 dev=2
libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
libusb: debug [linux_get_parent_info] Dev 0xfc06a0 (2-1) has parent 0xfc01d0 (usb2) port 1
libusb: debug [usbi_add_pollfd] add fd 6 events 1
libusb: debug [usbi_add_pollfd] add fd 8 events 1
libusb: debug [usbi_io_init] using timerfd for timeouts
libusb: debug [usbi_add_pollfd] add fd 10 events 1
libusb: debug [libusb_get_device_list] 
libusb: debug [discovered_devs_append] need to increase capacity
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_open] open 1.75
libusb: debug [usbi_add_pollfd] add fd 11 events 4
libusb: debug [libusb_set_configuration] configuration 1
libusb: debug [libusb_claim_interface] interface 0
Found HackRF board.
libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusb: debug [handle_events] poll() 4 fds with timeout in 60000ms
libusb: debug [handle_events] poll() returned 1
libusb: debug [reap_for_handle] urb type=2 status=0 transferred=1
[... 896 lines cut ...]
libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
libusb: debug [linux_get_parent_info] Dev 0xb476a0 (2-1) has parent 0xb471d0 (usb2) port 1
libusb: debug [usbi_add_pollfd] add fd 6 events 1
libusb: debug [usbi_add_pollfd] add fd 8 events 1
libusb: debug [usbi_io_init] using timerfd for timeouts
libusb: debug [usbi_add_pollfd] add fd 10 events 1
libusb: debug [libusb_get_device_list] 
libusb: debug [discovered_devs_append] need to increase capacity
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_open] open 1.75
libusb: debug [usbi_add_pollfd] add fd 11 events 4
libusb: debug [libusb_set_configuration] configuration 1
libusb: error [op_set_configuration] failed, error -1 errno 110
libusb: debug [libusb_close] 
libusb: debug [usbi_remove_pollfd] remove fd 11
hackrf_open() failed: HACKRF_ERROR_LIBUSB (-1000)
root@minerva:~#

-- 
Vennlig hilsen
Petter Reinholdtsen

dominicgs, 2016-10-31:
It appears that the HackRF is present in the final run, but when we try to access it, we are unable to set the configuration.  This still looks a lot like a power saving issue.

I don't know how else to investigate this.  I'll take a look at what other power saving measures have been added to USB stacks recently.  Are you running debian/ubuntu?

petterreinholdtsen, 2016-11-01:
[Dominic Spill]
> It appears that the HackRF is present in the final run, but when we
> try to access it, we are unable to set the configuration.  This still
> looks a lot like a power saving issue.

I'm happy to debug some more or update firmware, if someone can tell me
how to do it.

> I don't know how else to investigate this.  I'll take a look at what
> other power saving measures have been added to USB stacks recently.
> Are you running debian/ubuntu?

I'm running Debian Jessie (ie stable).
-- 
Happy hacking
Petter Reinholdtsen

miek, 2019-11-27:
As this is quite old and a lot has changed in the firmware & host software to address these issues, I'm going to close this. If you're still having trouble, please do open up a new issue.

closed 2019-11-27 by miek