greatscottgadgets/hackrf #244: 2.430891 ghz Bug

peterunlustig, 2016-03-29:
As shown in the video I get some strange signals at 2.430 ghz. The signal is random (so I don't think it is a real signal). It sometimes appears and sometimes it doesn't (no matter where I am). I can't produce anything by moving the usb cable/hackrf/antenna. 

Hackrf_info shows: Firmware Version: 2015.07.2

Is the hardware of my hackrf broken? Can I fix this somehow?

MFG Peter

[video.mp4.zip](https://github.com/mossmann/hackrf/files/193219/video.mp4.zip)

dominicgs, 2016-03-29:
This signal could be caused by many different things.  It looks to me like it could be a genuine signal, perhaps an alias of something nearby (this includes wifi).  Do you only see it at 2.43 GHz?  Or other frequencies?

If you use osmocom_fft to view the signal then you will be able to record a .cfile, which would be viewable in other tools, such as inspectrum.  I think this would be the first step to identify the signal.

peterunlustig, 2016-03-29:
Okay I will do that. Are there any signals like this for me it looked more like some interference? (On the screenshot it looks some kind of different) 
[image]

dominicgs, 2016-03-29:
You can try asking about the signal in the HackRF IRC channel, as someone else may recognise it.  You can also try http://www.sigidwiki.com/wiki/Signal_Identification_Guide

peterunlustig, 2016-03-29:
I just realized that while listening to music I get the signals like on the video. If I stop youtube though the signal changes into the one on the screenshot. Interestingly it just matters whether I am playing the music or not. Muting the sound doesn't change anything.

peterunlustig, 2016-03-29:
Okay so it is clearly some wifi interference. If I run music on its own, no strange signal. 
If I run some internet speedtest though the signal explodes.

This should be the solution. Thanks for help

schneider42, 2016-03-29:
The HackRF as an IF which can enter the WiFi region. If there are no shielding cans installed, WiFi can radiate into the IF stage and turn up as unwanted signals in the baseband. As the IF depends a bit on the chosen center frequency, this can be more evident at different center frequencies.

KC5CQW, 2016-03-29:
Nooelec has the shield BTW.

closed 2016-07-13 by dominicgs