greatscottgadgets/hackrf #245: No tx_callback until hackrf_stop_tx call if hackrf_start_tx called after receiving samples

vkchettimada, 2016-04-03:
I am receiving samples after which i use SIGINT to stop my main loop and then call hackrf_start_tx, as listed below. I dont get tx_callbacks until I ctrl-c again at which time I get some amount of callbacks.
I have removed printf's in the below code, but none of the API calls fail, hence wondering somethings wrong with pthread not running?

  retcode = hackrf_start_rx(p_device, sfn_hackrf_sample_block_cb, NULL);
  if (retcode)
  {
    goto exit_close;
  }

  do_exit = 0;
  while(!do_exit) {
    sleep(1);
  }

  retcode = hackrf_stop_rx(p_device);
  if (retcode)
  {
    goto exit_close;
  }

  while (hackrf_is_streaming(p_device) == HACKRF_TRUE);

  gs_mode = 1;

  retcode = hackrf_start_tx(p_device, sfn_hackrf_sample_block_cb, NULL);
  if (retcode)
  {
    goto exit_close;
  }

  do_exit = 0;
  while(!do_exit) {
    sleep(1);
  }

  retcode = hackrf_stop_tx(p_device);
  if (retcode)
  {
    goto exit_close;
  }

vkchettimada, 2016-04-03:
Pull request with fix.
https://github.com/mossmann/hackrf/pull/246

dominicgs, 2017-01-25:
Pull request was merged, so I'm closing this issue as it is fixed.

closed 2017-01-25 by dominicgs