greatscottgadgets/hackrf #248: Ubuntu install instructions are out of date

elliotcm, 2016-04-12:
Hi.

I'm setting up HackRF on a fresh Ubuntu 14.04LTS installation following the guide here:
https://github.com/mossmann/hackrf/wiki/Operating-System-Tips

At some point `app_store.py` has been removed, making all following instructions moot.

https://github.com/pybombs/pybombs (which is now gnuradio/pybombs) has install instructions which work but don't include instructions on setting up correct environment variables (at least as far as I can see).  This means that I have a prefix with lots of nice stuff in but none of the binaries can find their linked libraries.

If you can get me up to speed from here I'm happy to put in a pull request to update the documentation.

ckuethe, 2016-04-12:
Noted. I'll see about adding some more detail to the quickstart, and possibly an env generator.

For now, be advised that you will need to update your PATH, LD_LIBRARY_PATH, PKG_CONFIG_PATH, and PYTHONPATH. My setup_env.sh looks like this:

    R=/home/gnuradio
    
    export PATH=${GR}/bin:$PATH
    export LD_LIBRARY_PATH=${GR}/lib:$LD_LIBRARY_PATH
    export PKG_CONFIG_PATH=${GR}/lib/pkgconfig:$PKG_CONFIG_PATH
    export PYTHONPATH=${GR}/lib/python:${GR}/lib/python2.7/site-packages:${GR}/lib/python2.7/dist-packages:$PYTHONPATH

ckuethe, 2016-04-13:
https://github.com/gnuradio/pybombs/pull/300 brings back the ability to generate a setup_env.sh which you source to set up the environment variables correctly

elliotcm, 2016-04-13:
Awesome.  I haven't had a chance to populate the file myself yet but once I have I'll keep an eye on that PR and update the wiki / pybombs README when merged.

closed 2019-11-25 by miek