greatscottgadgets/hackrf #249: Does the HackRF actually support 2 Msps?

mhostetter, 2016-04-13:
I know there was discussion of decimating the 8 Msps down to 4 and 2 Msps in the CPLD.  I'm confused as to which rates are truly supported.

The Wiki says:
> supported sample rates: 2 Msps to 20 Msps (quadrature)

And hackrf_transfer.c says:

`printf("\t[-s sample_rate_hz] # Sample rate in Hz (4/8/10/12.5/16/20MHz, default %sMHz).\n",`

Thanks!

scateu, 2016-04-16:
It depends.

Based on the datasheet of MAX5864:

```
(Page 7) Note 2: The minimum clock frequency for the MAX5864 is 7.5MHz
```

But the MAX5864 chips have some design margin, it may work on some 
batches of MAX5864 while fail on other batches.

So it's not recommended to use lower sample rates directly.

You can add a rational resampler to do some sample rate conversion.

--
Wang Kang
Blog: http://scateu.me
Fingerprint: 011F 0492 97D6 5D75 8AC4  6458 D43F 3CE2 3353 B7BD
HAM Callsign: BH1RLW

mhostetter, 2016-04-16:
Ok.  Thanks for the reply.  

I guess I was asking because I saw lots of discussion previously in the Issues about decimating the 8 Msps to 4 Msps and maybe 2 Msps as well.  This would be done in the CPLD or microprocessor.  I was confused from the documentation which decimation rates have been implemented.

mhostetter, 2016-04-16:
Also because the chip seems to support baseband anti-alias filters of 1.75 MHz and 3.5 MHz, it seems an easy way to implement 2 Msps would be to configure the anti-alias filter to 1.75 MHz, ADC sample clock to 8 Msps, and then throw away 7 out of 8 samples.  Then you don't even have to filter in the CPLD/microprocessor.  The same would be true of 4 Msps using the 3.5 MHz analog filter setting.

Just a thought.

jboone, 2016-04-16:
@mhostetter This is exactly what the CPLD decimator is doing -- keeping one out of every N samples and discarding the rest. There isn't enough logic available in the CPLD to do any filtering, so it is performing "decimation" in a stricter sense of simply dropping samples. This technique does not remove out-of-band quantization noise during the downsampling, so does not provide the noise floor improvement one normally expects when oversampling.

All that said, I have stopped using the CPLD decimator, as running the MAX5864 at 2.5 to 4Msps works well enough in practice. I believe the main issue with running the MAX5864 below the recommended sampling rate is that the track-and-hold capacitor voltages will start to droop over such a long sampling interval, causing a small amount of distortion.

mossmann, 2016-04-16:
On Sat, Apr 16, 2016 at 10:13:19AM -0700, Jared Boone wrote:
> All that said, I have stopped using the CPLD decimator, as running the
> MAX5864 at 2.5 to 4Msps works well enough in practice. I believe the
> main issue with running the MAX5864 below the recommended [...]

I've never observed any distortion with 2 Msps ADC clock vs. 2 Msps by
decimating an 8 Msps or higher ADC clock, but it is theoretically
possible.  I've also never heard of any case where a MAX5864 failed to
operate at 2 Msps, so we changed the HackRF One specification to include
support for sample rates as low as 2 Msps.

The sample rates in the hackrf_transfer help are just suggestions.
Perhaps they should be removed if they are confusing.  Any rate from 2
Msps to 20 Msps is supported.  Rates below 2 Msps and above 20 Msps are
unsupported but can be specified and may work in some cases.  For
example, things seems to work reliably up to 21.5 Msps with my laptop.

Mike

mhostetter, 2016-04-16:
Oh!!  Ok, so I was confused.  I thought only an enumerated set of values were supported like 4, 8, 10, 12.5, 16, 20 MHz.

That's great!  Just discovered a whole new world! :)

jboone, 2016-04-16:
@mhostetter At one point (2013), only an enumerated set was supported, but code was added to compute fractional mode values for the Si5351C clock generator. See these commits (and maybe others) if you want the gory details:

1925649a01a289a2da2a25df501dd5d8f3296dce
f0c7fe66f13dd13e5021a782a913099141cf5926

Looks like we should treat this as an issue to remove the inaccurate and confusing text from the host tools. Perhaps this?

`# Sample rate in Hz (2MHz to 20MHz inclusive, default 10MHz).`

mhostetter, 2016-04-16:
That's great.  That's more functionality than I expected.  Also, that help message makes more sense to me.  

Thanks for the help guys.

closed 2016-05-30 by dominicgs