greatscottgadgets/hackrf #252: HackRF unable to transmit

sumeetjauhar, 2016-04-25:
I have a Pentoo linux distribution from which i am trying to use hackrf to tx and rx. I see that the 2 hackrf devices are correctly detected when i do a hackrf_info as shown below.

hackrf_info 
Found HackRF board 0:
Board ID Number: 2 (HackRF One)
Firmware Version: 2014.08.1
Part ID Number: 0xa000cb3c 0x006a434b
Serial Number: 0x00000000 0x00000000 0x14d463dc 0x0f54c1e1

Found HackRF board 1:
Board ID Number: 2 (HackRF One)
Firmware Version: 2014.08.1
Part ID Number: 0xa000cb3c 0x00544f5d
Serial Number: 0x00000000 0x00000000 0x14d463dc 0x2f3f5ce1

Unfortunately when i run my .grc file in gnuradio companion only one of these get detected ( as the rx) and the other gets an error (the tx) . Hence even though i can sniff packets i am unable to transmit anything. 

**_Using Volk machine: avx_32_mmx_orc
gr-osmosdr v0.1.4-48-g86ad5842 (0.1.5git) gnuradio 3.7.8
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf 
Using HackRF One with firmware 2014.08.1 
gr-osmosdr v0.1.4-48-g86ad5842 (0.1.5git) gnuradio 3.7.8
built-in sink types: uhd hackrf bladerf 
Number of USB devices: 12
USB device 1d50:6089:USB device 1d50:6089:
FATAL: Failed to open HackRF device (-5) HACKRF_ERROR_NOT_FOUND
_
Trying to fill up 1 missing channel(s) with null sink(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.**

I am attaching my osmocon source and sink configuration parameters just in case. 

Also i tried to explicitly specify the hackrf serial numbers in the osmocon sink but it does not work . How to fix this ?

[image]
[image]

dominicgs, 2016-04-25:
I believe that this is could have two possible causes.

1) Try specifying the HackRF that you want by serial number, replacing the hackrf=1 with hackrf=0f54c1e1

2) The other possible cause may be power management, which is already logged as #243.  The solution is to add exceptions for USB autosuspend in your system's power saving settings.  There are two different power saving systems for Linux, solutions for both can be found in the responses to #243.

sumeetjauhar, 2016-04-28:
Thanks for your suggestions but i tried to do what you suggested but no success .

1) Using serial number instead of the "1" in the osmocon sink argument does not work 

2) I went back to Ubuntu14.04 just so that i could get some support. So as suggested by you i downloaded the ubuntu laptop mode tools to disable the USB autosuspend. I did the following
 i) Created a blacklist for the hacrf device 
ii) Disabled the USB suspend feature entirely 
 
osboxes@osboxes:/etc/laptop-mode/conf.d$ cat usb-autosuspend.conf 
#
# Configuration file for Laptop Mode Tools module usb-autosuspend.
#
# For more information, consult the laptop-mode.conf(8) manual page.
#

###############################################################################
# USB autosuspend settings
# ------------------------
#
#__COMMENT If you enable this setting, laptop mode tools will automatically enable the
#__COMMENT USB autosuspend feature for all devices.
#__COMMENT
#__COMMENT NOTE: Some USB devices claim they support autosuspend, but implement it in a
#__COMMENT broken way. This can mean keyboards losing keypresses, or optical mice turning
#__COMMENT their LED completely off. If you have a device that misbehaves, add its USB ID
#__COMMENT to the blacklist below and complain to your hardware vendor.
#
################################################################################

# Enable debug mode for this module
# Set to 1 if you want to debug this module
DEBUG=0

# Enable USB autosuspend feature?
**# Set to 0 to disable
CONTROL_USB_AUTOSUSPEND="0"**

# Set this to use opt-in/whitelist instead of opt-out/blacklist for deciding
# which USB devices should be autosuspended.
# AUTOSUSPEND_USE_WHITELIST=0 means AUTOSUSPEND_*_BLACKLIST will be used.
# AUTOSUSPEND_USE_WHITELIST=1 means AUTOSUSPEND_*_WHITELIST will be used.
AUTOSUSPEND_USE_WHITELIST=0

# The list of USB IDs that should not use autosuspend. Use lsusb to find out the
# IDs of your USB devices.
# Example: AUTOSUSPEND_USBID_BLACKLIST="046d:c025 0123:abcd"
**AUTOSUSPEND_USBID_BLACKLIST="1d50:6089"
AUTOSUSPEND_USBID_BLACKLIST="1d50:604b"**

# The list of USB driver types that should not use autosuspend.  The driver
# type is given by "DRIVER=..." in a USB device's uevent file.
# Example: AUTOSUSPEND_USBID_BLACKLIST="usbhid usb-storage"
AUTOSUSPEND_USBTYPE_BLACKLIST=""

# The list of USB IDs that should use autosuspend. Use lsusb to find out the
# IDs of your USB devices.
# Example: AUTOSUSPEND_USBID_WHITELIST="046d:c025 0123:abcd"
AUTOSUSPEND_USBID_WHITELIST=""

# The list of USB driver types that should use autosuspend.  The driver
# type is given by "DRIVER=..." in a USB device's uevent file.
# Example: AUTOSUSPEND_USBTYPE_WHITELIST="usbhid usb-storage"
AUTOSUSPEND_USBTYPE_WHITELIST=""

# Trigger auto-suspension of the USB deivce under conditional circumstances
BATT_SUSPEND_USB=1
LM_AC_SUSPEND_USB=0
NOLM_AC_SUSPEND_USB=0

# USB Auto-Suspend timeout in seconds
# Number of seconds after which the USB devices should suspend
AUTOSUSPEND_TIMEOUT=2

Then i restart the laptop mode service / reboot the system both. 
However still i face the same problem of my second hackrf giving me errors

Using Volk machine: avx_32_mmx
gr-osmosdr v0.1.4-72-g164a09fc (0.1.5git) gnuradio v3.7.9.1-97-gc0381e0e
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy redpitaya 
Using HackRF One with firmware 2014.08.1 
gr-osmosdr v0.1.4-72-g164a09fc (0.1.5git) gnuradio v3.7.9.1-97-gc0381e0e
built-in sink types: uhd hackrf bladerf redpitaya file 
Number of USB devices: 4
USB device 1d50:6089:USB device 1d50:6089:
FATAL: Failed to open HackRF device (-5) HACKRF_ERROR_NOT_FOUND

Trying to fill up 1 missing channel(s) with null sink(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528

Some more information : 

gnuradio-companion --version
GNU Radio Companion v3.7.9.1-97-gc0381e0e

uname -a 
Linux osboxes 4.2.0-27-generic #32~14.04.1-Ubuntu SMP Fri Jan 22 15:32:27 UTC 2016 i686 i686 i686 GNU/Linux

Is there a way to test my Hackrf Tx ?

dominicgs, 2016-04-28:
I wasn't recommending that you install laptop-modes, I was saying that there are settings required if you already have it installed.  Have you checked for a TLP installation too?

Have you tried running your flowgraph with a file source and HackRF as the sink?  i.e. only one HackRF?

sumeetjauhar, 2016-04-28:
I checked my box , I dont have TLP installed on the box. 

Aplogies i am a newbie, What exactly do you mean by "a file source" . I can make use one of my HackRF's as the sink.

dominicgs, 2016-04-28:
Currently your flowgraph has one osmocom source and one osmocom sink.  I am suggesting that you replace the osmocom source block with a file or signal source.  They can be found in the list of blocks on the right hand side of gnuradio companion.  A signal source would probably be the easiest.

sumeetjauhar, 2016-04-28:
I did as you suggested i.e I replaced the input osmocom source with a Signal Source from the Gnuradio companion and kept the HackRF as the sink. 

I get the same error. I tried both hackrf=0 & hackrf=1 as device arguments in the osmocom sink. Does it mean that there is a problem in my sink parameter configuration?

dominicgs, 2016-04-28:
It certainly seems that way, but it could be a problem accessing the HackRF.  Could you try running `hackrf_transfer -t /dev/zero -f 2441000000 -s 4000000 -a 0 -x 0 -n 12000000` and see if it runs?

sumeetjauhar, 2016-04-29:
Seems that the command you suggested works. Also i see the Tx led of one of the HackRF glow up . 
```
sudo hackrf_transfer -c 64 -f 2441000000 -s 4000000 -a 0 -x 0 -n 12000000
call hackrf_sample_rate_set(4000000 Hz/4.000 MHz)
call hackrf_baseband_filter_bandwidth_set(3500000 Hz/3.500 MHz)
call hackrf_set_freq(2441000000 Hz/2441.000 MHz)
call hackrf_set_amp_enable(0)
samples_to_xfer 12000000/12Mio
Stop with Ctrl-C
 7.9 MiB / 1.000 sec =  7.9 MiB/second
 8.1 MiB / 1.001 sec =  8.1 MiB/second
 7.9 MiB / 1.001 sec =  7.9 MiB/second
 0.3 MiB / 1.000 sec =  0.3 MiB/second

User cancel, exiting...
Total time: 4.00153 s
hackrf_stop_tx() done
hackrf_close() done
hackrf_exit() done
exit
```
Now i want to test both my HackRF so i tried using the -d <serial number> option but i get a syntax error . Whats the correct way to force transfer via a specific hackrf ?

```sudo hackrf_transfer -d 2f7c35e1 -c 64 -f 2441000000 -s 4000000 -a 0 -x 0 -n 12000000 
Number of USB devices: 4
USB device 1d50:6089:USB device 1d50:6089:hackrf_open() failed: HACKRF_ERROR_NOT_FOUND (-5)
Usage:
 ```
gave me syntax error . I also tried with the ```0x``` before the serial number . 

I actually have a total of 4 HackRF's and i would be highly surprised if all of them are faulty. 

Can you see anything that catches your eye in the osmocom sink source device arguments i posted above ?

dominicgs, 2016-05-04:
Which version of the HackRF tools are you using?  It looks to me as though your firmware is from 2014.  Updating to the latest (2015.07) is required for serial number selection to work.

sumeetjauhar, 2016-05-09:
I updated my HackRF tools to the new firmware. I managed to get the hackrf_transfer utility to successfully transfer data using the ```hackrf_transfer -c 64 -f 2441000000 -s 4000000 -a 0 -x 0 -n 12000000``` . I also used the -d option to select the specific HackRF and i see the Tx LED glow. All this works fine,

However I still see issues in trying to make them work in sync i.e one Tx and other Rx. I get a different error now ( which on seeing the issues discussed in github seems to have been fixed )

```
Using Volk machine: avx_32_mmx
gr-osmosdr v0.1.4-72-g164a09fc (0.1.5git) gnuradio v3.7.9.1-97-gc0381e0e
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy redpitaya 
Using HackRF One with firmware 2015.07.2 
gr-osmosdr v0.1.4-72-g164a09fc (0.1.5git) gnuradio v3.7.9.1-97-gc0381e0e
built-in sink types: uhd hackrf bladerf redpitaya file 
Number of USB devices: 4
USB device 1d50:6089: 000000000000000014d463dc2f7612e1 match

FATAL: Failed to open HackRF device (-1000) HACKRF_ERROR_LIBUSB

Trying to fill up 1 missing channel(s) with null sink(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.
```
I can still Rx but not Tx. I tried explicitly specifying the serial no in the device arguments and still did not get it to work.

dominicgs, 2016-05-09:
Is this using the flowgraph from earlier in this thread?

How are you specifying the HackRF to use?  Serial number?  Numerical ID?

sumeetjauhar, 2016-05-09:
Yes i am using the flow graph mentioned earlier in this thread.
I specified the HackRF to be used in the Device arguments of the Osmocon Src and Sink as follows : 
i) hackrf=0, hackrf=1
ii) hackrf=0,serial=<serialno0> , hackrf=1,serial<serialno1>
iii)hackrf=<serialno0>,hackf=<serialno1>

All serial no without `0x`
All gave the same error .

cloatre, 2016-07-27:
Hello,
I have the same thing, one of my hackRf is not found:

---------------------------------------

root@kali:~/EZ-Wave/setup# hackrf_info 
Found HackRF board 0:
Board ID Number: 2 (HackRF One)
Firmware Version: 2014.08.1
Part ID Number: 0xa000cb3c 0x007e4755
Serial Number: 0x00000000 0x00000000 0x14d463dc 0x2f3b23e1

Found HackRF board 1:
USB descriptor string: 000000000000000014d463dc2f4375e1
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2
Part ID Number: 0xa000cb3c 0x0056475f
Serial Number: 0x00000000 0x00000000 0x14d463dc 0x2f4375e1

---------------------------------------

root@kali:~/EZ-Wave/setup# hackrf_transfer -c 64 -f 2441000000 -s 4000000 -a 0 -x 0 -n 12000000 -d 2f4375e1
Number of USB devices: 10
USB device 1d50:6089:USB device 1d50:6089: 000000000000000014d463dc2f4375e1 match
call hackrf_sample_rate_set(4000000 Hz/4.000 MHz)
call hackrf_baseband_filter_bandwidth_set(3500000 Hz/3.500 MHz)
call hackrf_set_freq(2441000000 Hz/2441.000 MHz)
call hackrf_set_amp_enable(0)
samples_to_xfer 12000000/12Mio
Stop with Ctrl-C
 7.9 MiB / 1.000 sec =  7.9 MiB/second
 8.1 MiB / 1.000 sec =  8.1 MiB/second
 7.9 MiB / 1.000 sec =  7.9 MiB/second
 0.3 MiB / 1.000 sec =  0.3 MiB/second

User cancel, exiting...
Total time: 4.00061 s
hackrf_stop_tx() done
hackrf_close() done
hackrf_exit() done
exit
root@kali:~/EZ-Wave/setup# hackrf_transfer -c 64 -f 2441000000 -s 4000000 -a 0 -x 0 -n 12000000 -d 2f3b23e1
Number of USB devices: 10
USB device 1d50:6089:USB device 1d50:6089: 000000000000000014d463dc2f4375e1 skip
hackrf_open() failed: HACKRF_ERROR_NOT_FOUND (-5)

----------------------------------------

lbb123, 2018-04-20:
hello,cloatre:
   I met the same question with you.
hackrf_info 
Found HackRF board 0:
USB descriptor string: 0000000000000000308066e62d35114b
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2
Part ID Number: 0xa000cb3c 0x006d4757
Serial Number: 0x00000000 0x00000000 0x308066e6 0x2d35114b
Found HackRF board 1:
USB descriptor string: 0000000000000000308066e622b5604b
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2
Part ID Number: 0xa000cb3c 0x00614764
Serial Number: 0x00000000 0x00000000 0x308066e6 0x22b5604b

int the gnuradio:
Executing: "/home/sdr/Desktop/grc/top_block.py"
linux; GNU C++ version 4.8.2; Boost_105400; UHD_003.005.005-0-unknown
Using Volk machine: avx_64_mmx_orc
gr-osmosdr 0.1.1 (0.1.1) gnuradio 3.7.2.1
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace 
Using HackRF One with firmware 2015.07.2 
gr-osmosdr 0.1.1 (0.1.1) gnuradio 3.7.2.1
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace 
FATAL: Failed to open HackRF device (-1000) HACKRF_ERROR_LIBUSB
Trying to fill up 1 missing channel(s) with null source(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.

But It still didn't work after settting "hackrf=2d35114b,hackrf=22b5604b",and the sps=1M.

anybody can help me?

dominicgs, 2018-04-23:
@lbb123  Where are you setting "hackrf=2d35114b,hackrf=22b5604b" ?  In the device arguments field?

You can't set multiple HackRF devices in one osmosdr block, it only supports one radio.

lbb123, 2018-04-24:
In one grc flow,I use two osmosdr blocks.one of them is sink,the other is source.In the device arguments field,I set them 'hackrf=2d35114b' and 'hackrf=22b5604b' respectively.But they didn't work synchronously.Only one of them can work. I still have this problem.

dominicgs, 2018-04-24:
Could you update your firmware and host code to the [2017 release](https://github.com/mossmann/hackrf/releases/tag/v2017.02.1)?

lbb123, 2018-04-25:
hello agin,dominicgs:
I think my firmware version is the latest "2015.07.2".By your advices, I try to update the host code but it didn't work as before.Could you help me?thank u

dominicgs, 2018-04-25:
You need to install the 2017 firmware too.  There are pre-compiled firmware binaries in the release archive.

SorenAndreasen, 2019-06-26:
and how do you install new firmware on windows? this seems impossible, or at least no information aroudn regarding this. cant find the hackrf tools for windows, a.n.y. where.

combatlord, 2019-06-27:
It ist possible in Windows7. I have Windows 7 64bit.
I have updated my hackrf to the latest FW without problems.
Dont forget to update the CPLD too.

Don't know if it works under Windows 10.

You can use the precompiled hackrf tools from photossdr Software.
This one are parts of the Software.
https://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D

closed 2019-11-27 by miek

Jorgmassih, 2020-10-22:
Setting the _Device Argument_ to `hackrf=<serialnumber in decimal>` worked for me.

yyyy8334051, 2021-03-19:
[image]

it has some problems that i cant figure out.
it show "Failed to open HackRF device (-5) HackRF not found" and my TX of hackrf one cant work
does someone can help me?

mohaissaoui, 2022-04-21:
relpace hackrf=0 with HackRF=0