greatscottgadgets/hackrf #253: Finding an OS to use HackRF

Fapricator, 2016-04-28:
I'm kind of stuck at the starting line and have had issues with getting started in an array of different Nix distros.

Pentoo RC4.6:
- known issue with USB booting. It hangs during the load and attempts to find it's own mount point
  Created with unetbootin and rufus
- Run from a burned disk. It loads but the installer fails. Same issue in a VM. x86 and x64
- Gentoo persistent USB drive instructions didn't work in Pentoo

Kali 2.0;
- It appears that GNU-Companion, GQRX and other resources were pulled from 2.0
- instruction to add Synaptic package manager all seem to point to old or missing repositories.
- I've added other repositories and attempted a handful of different tutorials to no avail.

Ubuntu 14.04 LTS:
- Issue installing Pybombs when it comes to app-store (this was already reported though)

Windows:
- HackRF works once I used the Airspy version of SDRSharp.
- If this is my only option...I'll look into building / porting tutorial... http://gnuradio.org/redmine/projects/gnuradio/wiki/WindowsInstall

I just found a Kali 1.0.5 iso that might work. I'm creating a bootable USB at the moment.

I'm really hoping that I've just overlooked something. I'm open to any suggestions.
Thanks!
Fapricator

markhellewell, 2016-04-28:
Did you try GNU Radio’s own Live DVD?  I haven’t yet, but it might be a dream to use…

http://gnuradio.org/redmine/projects/gnuradio/wiki/GNURadioLiveDVD

dominicgs, 2016-04-28:
I recommend using Ubuntu for SDR work because it's so widely used that most problems you are likely to come up against have already been solved.  The issue with app_store.py is a problem with the instructions on our wiki, but the new pybombs should work well without an app_store.

Detailed instructions are given in the [pybombs readme](https://github.com/gnuradio/pybombs/blob/master/README.md), but the following worked for me:
# Install pybombs
```
git clone https://github.com/gnuradio/pybombs.git
cd pybombs
python setup.py build
sudo python setup.py install
```
# Set a prefix
`sudo pybombs prefix init /usr/local -a globalprefix`
# Install GNURadio
`sudo pybombs install gnuradio gr-osmosdr`

sumeetjauhar, 2016-04-28:
Ubuntu 14.04 : 

@dominicgs  What would be the recommended way to install Hackrf . Is it via Pybombs too ? (Sorry to take over this issue instead of the one i created myself a couple of days ago since the question seems relevant here . )

I built HackRF from source on ubuntu 14.04 . The hackrf_info correctly showed the detected hackRF's . Now when i install gnuradio and gr0osmosdr via Pybombs and then later come back after installation to do a hackrf_info i get errors related to unresolved symbols :( Clearly I should not be building hackrf from source if i use Pybombs ?

dominicgs, 2016-04-28:
@sumeetjauhar I assume that you built HackRF from source using the latest release?

pybombs will install everything from the latest git head by default, so what you're probably seeing is a conflict between the last release that you have installed and the pybombs git install.  If you're going to use pybombs you would probably be best avoiding installing from source separately.

sumeetjauhar, 2016-04-28:
@dominicgs  yes correct. Thanks for the advice . I shall not build from source.

Fapricator, 2016-04-28:
@markhellewell Thanks! I just launched this from a USB drive. It's unfortunate that they don't have a persistence option yet but it appears to be in the works. I'll play with this more today and report back.

@dominicgs I'll give this a shot today. I appreciate you taking the time to write out the commands.

I'm going to try the suggested methods today. I appreciate everyone's quick and helpful responses!

dominicgs, 2016-07-13:
I am closing that, as I assume that you found a suitable OS.  If you would like to discuss OS options in the future, I think the [mailing list](https://pairlist9.pair.net/mailman/listinfo/hackrf-dev) would provide a wider discussion.

closed 2016-07-13 by dominicgs