greatscottgadgets/hackrf #255: Device re-connection and then can not work.

commthu, 2016-05-07:
I use hackrf as a receiver. The micro-usb cable is not stable,  the device may disconnect and re-connect when the device is receiving data,  when the re-connection occurs, my program can not  work cause it can not receive any data (rx_callback does not work), i have to shut down the program and re launch my program. Can you tell me another approach to solve this issue?

BTW, i have tried close the device and open the device,  sometimes can solve this issue, but most of the time fail, 

first, 
		result |= hackrf_stop_rx(device);
		result |= hackrf_close(device);
then,
		result = hackrf_init();
		result = hackrf_open(&device);

bg8wj, 2016-05-15:
same problem,I try to re-dfu and spi firmware use the new dfu bin code this problem still.

KC5CQW, 2016-05-16:
The cable is probably bent at an angle from excessive strain at the microUSB port. Possible cracked solder joints or a damaged jack. Investigate under high optical magnification (30x or more). If the hardware is OK, suspect weak power from the USB port. >500mA may be required. Check for power saving option settings on the host computer. Try a powered USB hub with an AC power supply ("wall-wart").

closed 2020-06-17 by miek