greatscottgadgets/hackrf #256: udev rules are incomplete

jdemel, 2016-05-13:
Hey!

Just figured out how I could get my HackRF One working.
The udev rules file that's shipped with this repo (pulled via pybombs this morning) says ... MODES=660 but after a little libusb digging and playing around it worked with MODES=0666
I got a LISBUSB -1000 error before.
I hope it helps to fix it.

Cheers
Johannes

dominicgs, 2016-05-13:
The udev rule makes assumptions about the group that your user belongs to when installing.  On some distributions this is not valid and you will find that the group that is given permission to access the HackRF is not suitable.  Changing from modes 660 to 0666 adds read/write access to the HackRF for everyone on the system.

While I'm glad this change works for you, I'm not sure it's the solution that I want to distribute to all HackRF users as it could cause problems on some systems.

jdemel, 2016-05-17:
I can see the problem here. It would be nice to have some guidance to track down the issue though. What I did:
Installed GNU Radio via PyBOMBS.
Tried to run hackrf_info -> failed
Run: LIBUSB_DEBUG=7 hackrf_info
-> libusb tells me "libusb requires write access to USB device nodes"
-> did some googling and came up with my fix.

Maybe the rules file could be edited with a few hints. Also, it might be helpful for users to get some info on the chosen config.

dominicgs, 2016-05-17:
If building from source, the cmake output contains information about which group is going to get the permissions.  Unfortunately pybombs suppresses this output and, depending on where you are installing to, doesn't install the rules file at all.

mossmann, 2017-01-27:
PyBOMBS has standardized on never installing udev rules.  This is going to continue to be a problem for some users, and they're just going to have to deal with udev manually.

closed 2017-01-27 by mossmann