greatscottgadgets/hackrf #257: I’m new to linux, compiling and that stuff.

GreenbeanDk, 2016-05-14:
Hi

_I installed Ubuntu-14.04.4-desktop-amd64.iso
Next I did this in termial window_
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install python-setuptools gr-osmosdr git git-core cmake libusb-1.0-0-dev build-essential libc6-dev libcppunit-1.13-0 libcppunit-dev libcppunit-subunit-dev libcppunit-subunit0 libgsl0-dev libgsl0ldbl python-cheetah g++ python-dev swig pkg-config libfftw3-dev libboost1.55-all-dev libcppunit-dev libgsl0-dev libusb-dev libsdl1.2-dev python-wxgtk2.8 python-numpy python-cheetah python-lxml doxygen libxi-dev python-sip libqt4-opengl-dev libqwt-dev libfontconfig1-dev libxrender-dev python-sip python-sip-dev
After that I made
$Mkdir /sdr
$cd  /sdr $ git clone https://github.com/mossmann/hackrf.git
$cd hackrf/host
$mkdir build && cd build
Now I in the build dir
$cmake ../ -DINSTALL_UDEV_RULES=ON _I get the error command no found_Which I presume is because the cmake is not found!
where do I go from here?

Best regards

Jens
Frederikshavn
Denmark

lukeweston, 2016-05-14:
Try cmake installation with: sudo apt-get install cmake

dominicgs, 2016-05-14:
As @lukeweston suggested, it doesn't look like cmake was installed.  Were there any error messages from the `apt-get install` step?

Do you get an error if you run `sudo apt-get install cmake` ?

GreenbeanDk, 2016-05-15:
HI
I've got the error after "cmake ../ -DINSTALL_UDEV_RULES=ON"
I made a fresh installtion and did all again from "sudo apt-get install python.."to"phyton-sip python-sip-dev" went ok
So at the moment I'm happy
Jens

dominicgs, 2016-05-15:
So the `cmake` command is now succeeding and you are able to run `make`?

GreenbeanDk, 2016-05-15:
Yes I did the make with success
Jens

dominicgs, 2016-05-15:
Thank you.  In that case, I'll close this issue as it is now working.

closed 2016-05-15 by dominicgs

GreenbeanDk, 2016-05-15:
I have an other qustion: shall install pybombs or gnu-radio? Are they the same or are two different things.
still nubi.

Dygear, 2016-05-15:
@GreenbeanDk, not really in the scope of the HackRF github repo. But I do agree with you that we could be more clear on how to install GNU Radio in general. As even the GNU Radio site doesn't include all of the steps needed to install PyBOMBs from start to finish.

To be honest, your best bet if you are like me, running 16.04 LTS as a fresh install is to use this script to install GNU Radio or simply `[sudo] apt-get gnuradio` or [`wget http://www.sbrac.org/files/build-gnuradio && chmod a+x build-gnuradio && ./build-gnuradio`](http://gnuradio.org/redmine/projects/gnuradio/wiki/InstallingGRFromSource)

GreenbeanDk, 2016-05-15:
Thank you Dygear I've try one of your suggestions
Jens