greatscottgadgets/hackrf #258: hackrf one can not work

bg8wj, 2016-05-14:
I got the new hackrf board and it work well with SDR v3.0 (windows ver), after one day work well it down, the tx light is still comes on , and then no data can be send to SDR. it can't work, when I click start looks well but no  data receive. also can stop with software "stop", only press the rest on board.
after I use the ubuntu 16 I reload the firmware 2015.7.2 as new board pross.. under the ubuntu, 
hackrf_info can got information well but still can't work. Is it the bug or  other ?

dominicgs, 2016-05-14:
I'm not sure which software you mean by "SDR v3.0" could you provide more details or a link to the software site?

Under Ubuntu you are able to run `hackrf_info`, do you see the 2015.07.2 firmware version listed?

Are you trying to transmit or receive?  You mention no data being sent, but I'm unsure if you mean from the radio to the software or from the software to the radio?

vsboost, 2016-05-14:
@dominicgs i think he is referring to Simon Brown's SDRConsole http://sdr-radio.com/v3_help as far as i am aware this is a receive only application.

KC5CQW, 2016-05-16:
Verify that you have a solid voltage on your USB port and that it is supplying 500mA or more of current. If you are on a laptop, it is possible that the current is too low for the HackRF. Some "Eco" settings will throttle down the USB when running on battery. The HackRF does have an on board regulator for voltage and some power will be lost (consumed) in the conversion/regulation. Try an AC powered USB hub between the computer and HackRF with a good 5Vdc output and >500mA current rating.

bg8wj, 2016-05-17:
Thanks everybody, after I check the clock source Si clock chip is down it just had the 9Hz clock output.
this is the problem of  it.

closed 2016-05-17 by bg8wj

abdullah612, 2017-01-16:
hello
 i have the same problem 
i have the HackRF One  running latest firmware 2015.07.02  it  was working until suddenly stopped
it is recognized by both hackrf_tools in VMware   and SDR# in windows 
but when i press the play button in SDR# the RX led lights but i get no IQ data  and when i press the stop button it hangs untill i reset the hackrf  
i did test it with hackrf_transfer  but i get the folowing error :

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 1.00091 s
hackrf_stop_rx() done
hackrf_close() failed: HACKRF_ERROR_LIBUSB (-1000)

i already checked the clock chip as @bg8wj suggested and it appears to be working 
i have 10MHz clock at CLKOUT   and a clock at the pins of ic equal to sample rate  and  double the sample rate all as documented in the schematic
i also checked voltages and they seem ok  at 1.8V and 3.3V
i also checked the RX_enable signal on pin 48 of the MAX2837 and it has 3.3V when i press start on SDR#

hackrf_info output : 
 
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2

my laptop core i7  16g ram