greatscottgadgets/hackrf #259: How to convert USB interface to Ethernet?

commthu, 2016-05-16:
How to convert USB interface to Ethernet? 
or do you have plan to change USB interface to Ethernet?

KC5CQW, 2016-05-16:
Use a raspberry pi running a tcp server for the HackRF data. 
Also run a tightVNC server for full "headless" remote administration. Be sure to provide 2.5A or more of available 5Vdc power to handle both boards. Provide solid EMI shielding between boards and place ferrite chokes on all cables (except RF coax, unless it is part of the antenna/feedline design).

commthu, 2016-05-16:
i will try，thank you very much！

发自我的 iPhone

dominicgs, 2016-06-14:
There are no plans to change the interface on HackRF One.  I think the best option for connecting to HackRF One via Ethernet is to use a single board computer as @KC5CQW suggested.  I personally prefer the BeagleBone Black to the Raspberry Pi, but almost any SBC should work.

closed 2016-06-14 by dominicgs